Ultimaker/CuraEngine #262: Compile Error, protobuf, not a member

w41ru5, 2015-10-12:
I am running Arch Linux 64bit and trying to compile the cura-build 15.10 repo, but I am running into a problem at CuraEngine. I have attached the output as a file due to the lengthy text, but it looks like it has something to do with protobuf. I am using google/protobuf compiled from comit 49f24af... Is there something else I am missing?

Ghostkeeper, 2015-10-13:
Ah, I wondered about this. Ultimaker's projects forked from protobuf a long time ago, because back then we wanted some things that Protobuf didn't offer at the time, such as CMake. Our version is [here](https://github.com/Ultimaker/protobuf).

Since then Protobuf has implemented many of those things itself. We've also made some bugfixes though (that should've been sent upstream maybe but weren't). It seems that Google's version isn't quite compatible with ours any more.

I suggest for now that you compile with [Ultimaker's fork of Protobuf](https://github.com/Ultimaker/protobuf).

Ghostkeeper, 2015-10-13:
Though it looks like this was made with the Cura-build repository.

awhiemstra, 2015-10-13:
cura-build normally builds the Ultimaker Protobuf clone of Protobuf. However, if you have Protobuf 3 locally installed CMake can get confused and use the local install of Protobuf for building Arcus/CuraEngine. If you can provide the full output, I can verify that that is what is happening here.

w41ru5, 2015-10-13:
Alright, I uninstalled all my protobuf compiles and reran the build and it went through fine. Thanks guys!

closed 2015-10-13 by w41ru5