Ultimaker/CuraEngine #263: Making 21474834 layers

Ghostkeeper, 2015-10-13:
CuraEngine just started overzealously creating 21474834 layers on my fairly small model, and then took extremely long to complete (it's currently working on skin). Meanwhile it also gave the message that it removed 21474834 layers because they are empty, but still takes long to complete the steps afterwards.

This is the model I'm slicing: http://dulek.net/work/cubecircle.stl

Just an observation, the maximum value for a signed integer is 2147483647. The number of layers I get is one thousandth of that, rounded up.

I'll post up the output of CuraEngine once it's done slicing.

Ghostkeeper, 2015-10-13:
This is the output of the program:

    Cura_SteamEngine version DEV
    Copyright (C) 2014 David Braam

    This program is free software: you can redistribute it and/or modify
    it under the terms of the GNU Affero General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    This program is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU Affero General Public License for more details.

    You should have received a copy of the GNU Affero General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
    Loading polygon setting machine_disallowed_areas not implemented...
    Unrecognized data type in JSON: machine_platform_offset has type array
    Loading polygon setting machine_head_polygon not implemented...
    Loading polygon setting machine_head_with_fans_polygon not implemented...
    Unrecognized data type in JSON: machine_head_with_fans_polygon has type array
    Unrecognized data type in JSON: machine_disallowed_areas has type array
    Unrecognized data type in JSON: machine_platform_offset has type array
    Loading /home/ruben/Models/planning/cube_circle.stl from disk...
    Failed to load model: /home/ruben/Models/planning/cube_circle.stl
    Unregistered setting mesh_position_z
    Unregistered setting mesh_position_y
    Unregistered setting mesh_position_x
    Unregistered setting center_object
    Loaded from disk in 0.009s
    Unregistered setting extruder_nr
     -g -e0 -s machine_heat_zone_length="16" -s machine_nozzle_expansion_angle="45" -s machine_nozzle_head_distance="3" -s machine_nozzle_tip_outer_diameter="1" -s machine_nozzle_size="0.4" -e0 -l "0"
    Unregistered setting support_extruder_nr
    Unregistered setting support_roof_extruder_nr
    Progress: start accomplished in 0.000s
    Starting slice...
    Slicing model...
    Layer count: 21474834
    Progress: slice accomplished in 43.470s
    Starting layerparts...
    Unregistered setting multiple_mesh_overlap
    Progress: layerparts accomplished in 99.660s
    Starting inset...
    Removing 21474834 layers because they are empty
    Progress: inset accomplished in 250.502s
    Starting support...
    Progress: support accomplished in 509.451s
    Starting skin...
    Unregistered setting ooze_shield_enabled
    Progress: skin accomplished in 288.078s
    Starting export...
    Unregistered setting machine_nozzle_offset_y
    Unregistered setting machine_nozzle_offset_x
    Unregistered setting machine_extruder_start_code
    Unregistered setting machine_extruder_end_code
    Unregistered setting switch_extruder_retraction_amount
    Unregistered setting switch_extruder_retraction_speed
    Unregistered setting switch_extruder_prime_speed
    Progress: export accomplished in 0.003s
    Total time elapsed 1192.79s.
    Print time: 0
    Filament: 0

    RUN FINISHED; exit value 0; real time: 19m 52s; user: 13s; system: 17m 40s

BagelOrb, 2015-10-14:
Isn't this solved now?

Ghostkeeper, 2015-10-15:
No, though it shouldn't be hard to put a check after the input to see whether the file exists. If none of the files provided exist it should give an error. If some of the files don't exist and some do, it should probably give a warning for each non-existent file.

Ghostkeeper, 2015-10-16:
Fixed in b91abb4.

closed 2015-10-16 by Ghostkeeper