Ultimaker/CuraEngine #264: No retract while leaving wipe tower.

karabas, 2015-10-13:
Dual head setup. Wipe and prime tower is enabled.
BUG: Massive stringing when move from the tower due to missing retract.

nickthetait, 2015-10-13:
I don't work for Ultimaker, but it is clear that more information is needed to solve a problem like this. Having a .gcode file with the specific location indicating the problem, pictures of printed parts, and name of your printer would all help tremendously.

daid, 2015-10-13:
And the version used also helps a lot.

BagelOrb, 2015-10-14:
The version you've used is the most important. Are you using the old Cura? Or the new one (15.06 or 15.10 beta)?

karabas, 2015-10-14:
Cura 15.04.02. 
printer cartesian(prism uni) modified for use with single E3D hotend and mixer for two bowdens.
Cura settings are for dual head setup and wipe&prime tower=90mm3.
Printing is good except of strings between tower and model. Retract inside model parts is ok.

BagelOrb, 2015-10-14:
Dual extrusion is not fully supported yet - especially not in the old Cura.

We will not solve the bug in the old Cura - you will have to wait for the new Cura to support dual extrusion.

closed 2015-10-14 by BagelOrb