Ultimaker/CuraEngine #265: 15.09.90: "Connect Zig Zags" Setting Not Functioning

CCS86, 2015-10-15:
I am seeing the support zigzags connected regardless of this setting's state. Is anyone else seeing this?

[image]

BagelOrb, 2015-10-15:
Perhaps the description is a bit unclear.

Normal ZigZags will look like 
``` 
..|""|..|""|..
```
While with Connect ZigZags they should look like
```
::|""|..|""|::
```
connected:
[image]
not connected:
[image]

On the right hand side of the triangle the setting seems to do what it's supposed to.

However, on the left hadn side of the triangle it doesn't seem to connect the end piece back to the rest.

BagelOrb, 2015-10-15:
The goal of the setting is to start on one of those straight scanlines instead of starting on the outer tip of the support, cause sometimes the outer tip starts in mid air.

CCS86, 2015-10-15:
Ah, I gotcha! 

What I had in my head, was probably more like "lines" vs "zig zags"

BagelOrb, 2015-10-22:
There's still a bug: the top picture shows a newly introduced line (bottom left) - away from the already connected lines, while the whole purpose was to get the top left connected back to the bottom left again.

The line piece from the top left to the bottom left is missing.

I think this is caused by the fact that that line is in the same direction as the vertical scanlines.

BagelOrb, 2015-12-16:
Closing this since the new bug report #287 is more accurate.

closed 2015-12-16 by BagelOrb