Ultimaker/CuraEngine #267: cross-compiling for openwrt mips and x86 (uClibc build)

artynet, 2015-10-21:
Hello folks,

I have almost succedeed in cross-compiling CuraEngine for OpenWRT since I have made the port of OctoPrint for such distro. I have successfully cross-compiled protobuf v3 beta1 and the latest release of libArcus but I am having a build error when working with the missing piece ! Here's the output error : 

```
/home/arturo/Scaricati/openwrt-15.05-x86/build_dir/target-i386_i486_uClibc-0.9.33.2/CuraEngine-15.06.03/src/main.cpp: In function 'int main(int, char**)':
/home/arturo/Scaricati/openwrt-15.05-x86/build_dir/target-i386_i486_uClibc-0.9.33.2/CuraEngine-15.06.03/src/main.cpp:105:32: error: 'stoi' is not a member of 'std'
                         port = std::stoi(ip_port.substr(ip_port.find(':') + 1).data());
                                ^
make[5]: *** [CMakeFiles/CuraEngine.dir/src/main.cpp.o] Error 1
make[5]: Leaving directory `/home/arturo/Scaricati/openwrt-15.05-x86/build_dir/target-i386_i486_uClibc-0.9.33.2/CuraEngine-15.06.03'
make[4]: *** [CMakeFiles/CuraEngine.dir/all] Error 2
make[4]: Leaving directory `/home/arturo/Scaricati/openwrt-15.05-x86/build_dir/target-i386_i486_uClibc-0.9.33.2/CuraEngine-15.06.03'
make[3]: *** [all] Error 2
make[3]: Leaving directory `/home/arturo/Scaricati/openwrt-15.05-x86/build_dir/target-i386_i486_uClibc-0.9.33.2/CuraEngine-15.06.03'
make[2]: *** [/home/arturo/Scaricati/openwrt-15.05-x86/build_dir/target-i386_i486_uClibc-0.9.33.2/CuraEngine-15.06.03/.built] Error 2
make[2]: Leaving directory `/home/arturo/Scaricati/openwrt-15.05-x86/package/linino/curaengine'
make[1]: *** [package/linino/curaengine/compile] Error 2
make[1]: Leaving directory `/home/arturo/Scaricati/openwrt-15.05-x86'
make: *** [package/curaengine/compile] Errore 2
```

I have also previously applied this patch :

```
--- a/src/main.cpp  2015-08-05 14:30:57.000000000 +0200
+++ b/src/main.cpp  2015-10-21 18:12:06.268403637 +0200
@@ -4,8 +4,10 @@
 #include <string.h>
 #include <sys/time.h>
 #include <signal.h>
-#if defined(__linux__) || (defined(__APPLE__) && defined(__MACH__))
+#ifndef __UCLIBC__
 #include <execinfo.h>
+#endif
+#if defined(__linux__) || (defined(__APPLE__) && defined(__MACH__))
 #include <sys/resource.h>
 #endif
 #include <stddef.h>
```
to prevent the "can't find execinfo.h" header error. By googling around it seems that patch like this one are the usual modification to the source file to fix this issue...any pointers about the error anyway ?

   Kind Regards, Arturo

daid, 2015-10-21:
Your compiler is outdated. std::stoi is part of C++11
http://www.cplusplus.com/reference/string/stoi/

artynet, 2015-10-22:
I bump into the same error with gcc v4.8.3 and v5.2.0...maybe some missing configuration options when buildgin the toolchain ?

daid, 2015-10-22:
```
gcc --version
gcc (Debian 4.9.2-10) 4.9.2
```

awhiemstra, 2015-10-22:
Which C++ standard library are you using? Because that would be the library defining std::stoi.

artynet, 2015-10-22:
I think I have found the reason...as @awhiemstra has suggested **uClibc** doesn't support the std=c++11 set of instructions (it is the embedded library I'm currently using). The only solution is to switch to the **musl** embedded which support it. To test it, I have simpy compiled the basic example provided from @daid to me in the second post...

daid, 2015-10-22:
Well, depending on your use, you might not need the line that is giving the compile issues and can just comment it.

Ghostkeeper, 2015-10-23:
All of libArcus is only used to communicate with the front-end framework, so if the program is only to be used from the command line it can be omitted. Removing all calls to Arcus in commandSocket.cpp and removing it from CMakeLists.txt should suffice.

awhiemstra, 2015-10-23:
Except that the engine also uses C++11 in other places as far as I know. If uClibc does not support C++11 you will run into issues in those other places.

artynet, 2015-10-26:
I see. The built went smoothly when building  with **musl** my OpenWRT custom image. 

It was in my plans to switch to **musl** anyway since uClibc is quite oudated and the future releases of OpenWRT, of course, will support **musl** itself as default C embdedded library. All of you have been very helpful to me. I think we can close the issue now.

Kind Regards, Arturo

closed 2015-10-26 by artynet