Ultimaker/CuraEngine #269: Error code -8 on any model

Renha, 2015-10-24:
I am trying to slice any model using OctoPrint. Have fixed syntax and version detection in local copy of Octoprint... to get that. No explaination what's going on, no CuraEngine version in output (it supposed to be last), nothing useful - and unknown error:

```
2015-10-24 12:58:21,680 ### Slicing /home/root/.octoprint/uploads/bedholder.stl to /tmp/tmpAMrT37.gco using profile stored at /tmp/slicing-profile-temp-jsX7zE.profile
2015-10-24 12:58:22,565 
2015-10-24 12:58:22,567 Cura_SteamEngine version DEV
2015-10-24 12:58:22,569 Copyright (C) 2014 David Braam
2015-10-24 12:58:22,571 
2015-10-24 12:58:22,573 This program is free software: you can redistribute it and/or modify
2015-10-24 12:58:22,589 it under the terms of the GNU Affero General Public License as published by
2015-10-24 12:58:22,591 the Free Software Foundation, either version 3 of the License, or
2015-10-24 12:58:22,593 (at your option) any later version.
2015-10-24 12:58:22,595 
2015-10-24 12:58:22,597 This program is distributed in the hope that it will be useful,
2015-10-24 12:58:22,599 but WITHOUT ANY WARRANTY; without even the implied warranty of
2015-10-24 12:58:22,601 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2015-10-24 12:58:22,603 GNU Affero General Public License for more details.
2015-10-24 12:58:22,605 
2015-10-24 12:58:22,607 You should have received a copy of the GNU Affero General Public License
2015-10-24 12:58:22,609 along with this program.  If not, see <http://www.gnu.org/licenses/>.
2015-10-24 12:58:22,643 Loading /home/root/.octoprint/uploads/bedholder.stl from disk...
2015-10-24 12:58:22,811 Unregistered setting machine_extruder_count
2015-10-24 12:58:22,812 Unregistered setting mesh_position_z
2015-10-24 12:58:22,814 Unregistered setting mesh_position_y
2015-10-24 12:58:22,816 Unregistered setting mesh_position_x
2015-10-24 12:58:22,818 Unregistered setting center_object
2015-10-24 12:58:22,820 Loaded from disk in 0.195s
2015-10-24 12:58:22,821 Unregistered setting extruder_nr
2015-10-24 12:58:22,823 -g -e0 -l "0"
2015-10-24 12:58:22,825 Unregistered setting support_extruder_nr
2015-10-24 12:58:22,827 Unregistered setting support_roof_extruder_nr
2015-10-24 12:58:22,828 Progress: start accomplished in 0.000s
2015-10-24 12:58:22,830 Starting slice...
2015-10-24 12:58:22,832 Slicing model...
2015-10-24 12:58:22,834 Unregistered setting meshfix_keep_open_polygons
2015-10-24 12:58:22,836 Unregistered setting meshfix_extensive_stitching
2015-10-24 12:58:22,838 Unregistered setting magic_mesh_surface_mode
2015-10-24 12:58:22,965 Progress:slice:1:1 	0.018531%
2015-10-24 12:58:22,967 Layer count: 56
2015-10-24 12:58:22,975 Progress: slice accomplished in 0.163s
2015-10-24 12:58:22,977 Starting layerparts...
2015-10-24 12:58:22,979 Unregistered setting multiple_mesh_overlap
2015-10-24 12:58:22,983 Unregistered setting meshfix_union_all_remove_holes
2015-10-24 12:58:22,987 Unregistered setting meshfix_union_all
2015-10-24 12:58:22,990 Progress:layerparts:1:1 	0.023922%
2015-10-24 12:58:22,993 Progress: layerparts accomplished in 0.055s
2015-10-24 12:58:22,996 Starting inset...
2015-10-24 12:58:22,998 Unregistered setting remove_overlapping_walls_x_enabled
2015-10-24 12:58:23,000 Unregistered setting remove_overlapping_walls_0_enabled
2015-10-24 12:58:23,002 Progress:inset:1:56 	0.025363%
2015-10-24 12:58:23,007 Progress:inset:2:56 	0.026805%
2015-10-24 12:58:23,015 Progress:inset:3:56 	0.028246%
[cut]
2015-10-24 12:58:23,345 Progress:inset:54:56 	0.101762%
2015-10-24 12:58:23,349 Progress:inset:55:56 	0.103204%
2015-10-24 12:58:23,352 Progress:inset:56:56 	0.104645%
2015-10-24 12:58:23,357 Progress: inset accomplished in 0.342s
2015-10-24 12:58:23,359 Starting support...
2015-10-24 12:58:23,361 Unregistered setting support_roof_enable
2015-10-24 12:58:23,363 Progress: support accomplished in 0.002s
2015-10-24 12:58:23,365 Starting skin...
2015-10-24 12:58:23,367 Unregistered setting skin_no_small_gaps_heuristic
2015-10-24 12:58:23,369 Progress:skin:1:56 	0.287107%
[cut]
2015-10-24 12:58:23,428 Progress:skin:9:56 	0.311653%
2015-10-24 12:58:23,433 Progress:skin:10:56 	0.314722%
2015-10-24 12:58:23,446 Progress:skin:11:56 	0.317790%
[cut]
2015-10-24 12:58:24,071 Progress:skin:55:56 	0.452798%
2015-10-24 12:58:24,078 Progress:skin:56:56 	0.455867%
2015-10-24 12:58:24,081 Unregistered setting infill_sparse_combine
2015-10-24 12:58:24,083 ### Finished, returncode -8
2015-10-24 12:58:24,087 ----------------------------------------
```

other model i've tried have same returncode on 66% or similar

Renha, 2015-10-24:
```
2015-10-24 12:08:37,461 Command called:
2015-10-24 12:08:37,463 /home/root/newcura/CuraEngine/build/CuraEngine slice -v -p -s coolHeadLift=0 -s downSkinCount=3 -s enableCombing=1 -s endCode=M104 S0                     ;extruder heater off
2015-10-24 12:08:37,475 M140 S0                     ;heated bed heater off (if you have it)
2015-10-24 12:08:37,477 G91                                    ;relative positioning
2015-10-24 12:08:37,479 G1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure
2015-10-24 12:08:37,481 G1 Z+0.5 E-5 X-20 Y-20 F12000 ;move Z up a bit and retract filament even more
2015-10-24 12:08:37,494 G28											 ;move X/Y to min endstops, so the head is out of the way
2015-10-24 12:08:37,496 G4 S10
2015-10-24 12:08:37,498 M84                         ;steppers off
2015-10-24 12:08:37,499 G90                         ;absolute positioning
2015-10-24 12:08:37,501 -s extrusionWidth=400.0 -s fanFullOnLayerNr=2 -s fanSpeedMax=0 -s fanSpeedMin=0 -s filamentDiameter=1750 -s filamentFlow=100 -s fixHorrible=1 -s gcodeFlavor=0 -s infillOverlap=20 -s infillSpeed=50 -s initialLayerSpeed=20 -s initialLayerThickness=200 -s initialSpeedupLayers=4 -s inset0Speed=30 -s insetCount=2 -s insetXSpeed=30 -s layer0extrusionWidth=400 -s layerThickness=200 -s minimalExtrusionBeforeRetraction=20 -s minimalFeedrate=10 -s minimalLayerTime=10 -s moveSpeed=200 -s multiVolumeOverlap=150 -s objectSink=0 -s postSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.
2015-10-24 12:08:37,503 ;This code is added after the T(n)
2015-10-24 12:08:37,511 -s posx=0.0 -s posy=0.0 -s preSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.
2015-10-24 12:08:37,513 ;This code is added before the T(n)
2015-10-24 12:08:37,515 -s printSpeed=100 -s retractionAmount=5400 -s retractionAmountExtruderSwitch=16500 -s retractionMinimalDistance=1500 -s retractionSpeed=100 -s retractionZHop=200 -s skirtDistance=3000 -s skirtLineCount=2 -s skirtMinLength=150000 -s sparseInfillLineDistance=1333 -s startCode=M140 S90.0
2015-10-24 12:08:37,517 M109 T0 S235.0
2015-10-24 12:08:37,518 T0
2015-10-24 12:08:37,520 M190 S90.0
2015-10-24 12:08:37,522 
2015-10-24 12:08:37,523 ;Basic settings: Layer height: 0.2 Walls: 0.8 Fill: 30
2015-10-24 12:08:37,525 ;Print time: ?print_time?
2015-10-24 12:08:37,527 ;Filament used: ?filament_amount?m ?filament_weight?g
2015-10-24 12:08:37,528 ;Filament cost: ?filament_cost?
2015-10-24 12:08:37,530 ;M190 S90 ;Uncomment to add your own bed temperature line
2015-10-24 12:08:37,532 ;M109 S235 ;Uncomment to add your own temperature line
2015-10-24 12:08:37,534 G21        ;metric values
2015-10-24 12:08:37,536 G90        ;absolute positioning
2015-10-24 12:08:37,538 M82        ;set extruder to absolute mode
2015-10-24 12:08:37,539 M107       ;start with the fan off
2015-10-24 12:08:37,541 G28			;Re-home
2015-10-24 12:08:37,543 G1 Z15.0 F12000 ;move the platform to 15mm
2015-10-24 12:08:37,544 G92 E0                  ;zero the extruded length
2015-10-24 12:08:37,546 G1 F200 E3              ;extrude 3mm of feed stock
2015-10-24 12:08:37,548 G92 E0                  ;zero the extruded length again
2015-10-24 12:08:37,549 G1 F12000
2015-10-24 12:08:37,551 ;Put printing message on LCD screen
2015-10-24 12:08:37,553 M117 Printing...
2015-10-24 12:08:37,554 -s supportAngle=-1 -s supportEverywhere=0 -s supportExtruder=-1 -s supportLineDistance=2000 -s supportXYDistance=800 -s supportZDistance=150 -s upSkinCount=3 -o /tmp/tmpiE1MpP.gco -l /home/root/.octoprint/uploads/bedholder.stl
2015-10-24 12:08:37,556 
```

BagelOrb, 2015-10-26:
CuraEngine is called with wrong arguments.

The settings are parsed by using the space as separator; the end gcode contains spaces and is not parsed correctly. The end gcode should be surrounded by quotation marks.

Also with the new CuraEngine the setting names have changed. For example: `downSkinCount` is now `top_layers`.

Please contact OctoPrint.

Renha, 2015-10-26:
still
>No explaination what's going on, no CuraEngine version in output, nothing useful

Renha, 2015-10-26:
thank you for explain, but CuraEngine's error output must do it, not you :-/

daid, 2015-10-26:
You should be using the legacy branch with octoprint (most likely)

BagelOrb, 2015-11-04:
We're working on the CuraEngine version number. Currently it shouldn't be possible (any more) to exit with an error code without a message.

If you do find a way to exit with error code but without error message, please provide me with the command line arguments to the program and the STL you tried to slice.

closed 2015-11-04 by BagelOrb