Ultimaker/CuraEngine #271: [15.09.91] Engine crashes when layer height is set to anything above .1

mikenew12, 2015-10-27:
Happens even with completely default settings. Slicing works fine at .1mm layer height, but at .11 (or anything beyond .1) the engine crashes. The log just says "Backend crashed or closed. Restarting..." Rolling back to 15.09.90 fixes the issue. I'm on OS X 10.11.1, but the same issue also happens on Windows 7.

I'm not sure how to get debug output from the engine itself, but if you need it let me know and I'll figure it out.

Ghostkeeper, 2015-10-28:
This one was recently fixed. The problem was that there was a setting to determine the layer thickness of infill. This setting is rounded down to the next layer height, since the engine does this by combining a number of layers (which must be integer).

The fix was to make the infill layer thickness inherit from the normal layer thickness if the setting is not user-set. Also, the engine now always uses a minimum of 1 layer. This should be included in the upcoming build next Friday.

Until then you can manually set the infill layer thickness to be at least as thick as the layer height, as a workaround. Or you can use the 15.10 branches from the Uranium, Cura and CuraEngine repositories if you can build it yourself.

closed 2015-10-28 by Ghostkeeper

mikenew12, 2015-10-28:
Thanks, I appreciate the explanation and the workaround. I haven't been able to build from source in about a month or so, but maybe I'll take another look at that.