Ultimaker/CuraEngine #272: Machine overrides are ignored

Renha, 2015-10-29:
I am trying to create machine file for Rostock MAX, but it's overrides are ignored.
here is my file: http://pastebin.com/XejQqjTx
Running with command:
`CuraEngine slice -v -j rmax.json -l ddashie.stl -o ddashie.gcode`
Then i look into generated gcode and there I can see wrong temperatures, layer thicknesses, etc - but correct end/start gcode.

Renha, 2015-10-29:
i've tried with that strange lines, copied from somewhere
```
    "machine_extruder_trains": [{
        "machine_nozzle_size": {
            "default": 0.4
        },
        "machine_nozzle_heat_up_speed": { 
            "default": 2.0 
        },
        "machine_nozzle_cool_down_speed": { 
            "default": 2.0 
        },
        "machine_nozzle_tip_outer_diameter": {
            "default": 1
        },
        "machine_nozzle_head_distance": {
            "default": 3
        },
        "machine_nozzle_expansion_angle": {
            "default": 45
        },
        "machine_heat_zone_length": { 
            "default": 16
        }
    }],
```
then no error but still no influence of "override" category

BagelOrb, 2015-10-30:
This seems like a CuraEngine issue.

Note that this isn't an issue for the Cura frontend, since settings are passed via the command socket, not as command line arguments.

What "strange lines" are you talking about?

BagelOrb, 2015-10-30:
fixed in next release.

Renha, 2015-10-30:
Thank you. About strange lines: These lines are there in some machine files, and are missing in another as I can remember.

>Note that this isn't an issue for the Cura frontend

That's why I have posted it there, not in Cura issues :) and I'm unable to get working Cura in Arch Linux and Sailfish OS, so I have only CLI for now.

BagelOrb, 2015-11-02:
Can you please specify which lines are the strange ones?

All overrides should be present in the parent json file and all machine settings should be present in each machine json settings file.

So do you mean you found machine settings which are not uniformly present over all machine json files?

Renha, 2015-11-02:
Yes! https://github.com/Ultimaker/Cura/blob/master/resources/machines/prusa_i3.json in last revision of Cura, and many others.

I wish to have example settings file, always correct one, with full description of every option - at the moment, for example, I don't know what is "machine_nozzle_heat_up_speed" - is it to heat up at 1 celsius, or 1 farenheit, or from room temperature to 100 celsius, or how fast heating process is moved over hotend in mm/s. If that's explained somewhere, it would be great to have a link at least - but example up-to-date file would be much better.

Thank you!

BagelOrb, 2015-11-02:
@Renha, I wish to know ***which settings*** are not present in some files while they are in others. Which setting did you see in other json files which isn't present in prusa_i3.json?

'machine_nozzle_heat_up_speed' is an estimate of the average speed in *C/s the nozzle heats up within the bounds of normal printing temperatures. So you could use the average time it takes to heat up from 180 to 230 and use the data to compute the value.

Indeed we should add descriptions to the machine settings as well.

(closing this since the original issue has been solved)

closed 2015-11-02 by BagelOrb

Renha, 2015-11-02:
sorry for misunderstooding,my english is bad.

 "machine_extruder_trains" isn't present in prusa_i3.json, and when I remove that settings from my own rmax.json, I've got:
>Error: no machine_extruder_trains category found in JSON!

BagelOrb, 2015-11-03:
machine_extruder_trains was introduced in order to support dual extrusion. For every extruder train (combination of feeder, bowden tube, nozzle) you supply the default settings in the json.

However, if viewed as such, single extrusion is a special case of multi-extrusion, with extruder_count=1.

I've changed the code: now it just doesn't generate an error and excepts json files with no extruder trains defined. Given that machine_extruder_count is set to 1 by default in fdmprinter.json, this means that a profile with no extruder trains defined is the same as a profile with an empty extruder train.
If no settings are provided in an extruder train, then all extruder train settings are inherited from the global settings.

I've also added descriptions to the machine settings.