Ultimaker/CuraEngine #273: Fill_sparse_density working?

Limseunghwan, 2015-10-29:
Hello 
I am trying to create my own settings for Cura Engine.
during my works i found some strange point and leave a message.

I use curaEngine.exe, some dll files and fdmprinter.json file in Cura 15.06.03 version
which can download from this link(https://ultimaker.com/en/cura-software/list)

I tried 

CuraEngine -v -j fdmprinter.json -o test.gcode robot.stl
and make test.gcode file.

Then I changed default value of fill_sparse_density in fdmprinter.json (20 -> 80)
But it`s not working. There are no change in test.gcode file

However if I change other values same way(fill_pattern, fill_overlap .....) then 
test.gcode is changing

So I tried to search this part in github at branch 15.06 and 15.10
(https://github.com/Ultimaker/CuraEngine/search?utf8=%E2%9C%93&q=fill_sparse_density)
but it looks it works nothing.

Is there any way to change fill_sparse_density?

Thank you for reading.

Ghostkeeper, 2015-10-30:
Note that in branch 15.10, this setting was renamed to `infill_sparse_density`.

What you're looking at is a bit messy (I think). The `fill_sparse_density` setting is not read by the engine at all. It is used only in the front-end at the moment. Most importantly, the front-end computes the `infill_line_distance` setting from it with the formula `0 if fill_sparse_density == 0 else (infill_line_width * 100) / fill_sparse_density`. Also, if `fill_sparse_density` is more than 25, the default infill pattern becomes grid.

So to fix it you would have to supply the appropriate value for `infill_line_distance`.

BagelOrb, 2015-10-30:
Generally speaking only the settings which have no children in the json file are used by the engine.

closed 2015-10-30 by BagelOrb

Limseunghwan, 2015-10-30:
@Ghostkeeper  , @BagelOrb 
Thank you for your reply.
I change infill_line_distance and get result what I want to expect.

Have a nice halloween!

mikemeding, 2016-02-25:
@Ghostkeeper 
I cannot tell you how long I have been trying to figure that out. 
Thanks so much!

TopCoderKZ, 2017-05-11:
@Limseunghwan how did you change infill_sparse_density using infill_line_distance? Please help.

Ghostkeeper, 2017-05-12:
If you change the line distance, the lines will be further apart so the infill becomes less dense.

Only the line distance is actually used. Normally the line distance is calculated from the density and the line thickness.