Ultimaker/CuraEngine #275: Missed inner shell loops

CCS86, 2015-11-21:
This is a bummer. You can see that I have this print tuned towards over-extrusion, because it needs to be functionally strong. There are two 8mm holes. One is being filled correctly and the other is not:

[image]

[image]

[UMO_Plate e top.gcode.pdf](https://github.com/Ultimaker/CuraEngine/files/40742/UMO_Plate.e.top.gcode.pdf)

CCS86, 2015-11-21:
11-17 Build

daid, 2015-11-21:
Are you sure the model is without errors? As this usually happens when there are extra internal surfaces (for example, this happens quite often with Sketchup models)

CCS86, 2015-11-21:
It is always possible. I am using a nice parasolid modeller though. Here's the model:

[Plate e bot.stl.pdf](https://github.com/Ultimaker/CuraEngine/files/40902/Plate.e.bot.stl.pdf)

CCS86, 2015-11-21:
I called for 10 perimeter laps on this print, and interestingly, the gap appears to be perfectly centered within them.

Also noteworthy, I ran with a commanded line width of .5mm (on my .4mm nozzle), and .6mm infill line width. The rest of the print seemed to handle the extrusion amount and spacing well.

BagelOrb, 2015-11-23:
The model shows no problems in X-ray view.

BagelOrb, 2015-11-23:
The perimeters are all in the gcode, so I guess this is just a fluke. 

Please contact us if it happens repeatedly on this gcode.

PS:
I'm curious: why did you set the line widths to amounts differing from the nozzle size?

closed 2015-11-23 by BagelOrb

CCS86, 2015-12-10:
I think that going just a bit over the nozzle nominal size helps lay down more of the ideal rectangular cross section. There is a pretty good chance that most nozzles are slightly oversized between machining tolerance and wear. Plus there is a flat bordering the nozzle orifice to handle the overshoot. 

Also, on infill, it can be much more efficient for a given layer size.

BagelOrb, 2015-12-10:
Hmmm. I think I heard that the tolerances are greater in the inward direction. I've heard people having more success by setting the nozzle size to 0.35...

CCS86, 2015-12-10:
Just from a machinists point of view, if you take a 0.4mm drill, and poke a hole in a relatively soft metal, it will be slightly larger than 0.4mm. Then add in cumulative wear, which will continue to increase the orifice size over time, and I'm not sure how it could be smaller.

I have not done much testing with defining smaller-than-nominal nozzle size yet to compare. Worth a try for sure.

Just as a thought experiment, my intuition says that the closer your layer height : nozzle width ratio gets to 1, the harder it will be to connect adjacent beads securely. It makes me want to try say: 0.35mm width for outer perimeters, 0.45mm for other perimeters and 0.55-0.6mm for infill as a balance of skin detail, and overall print strength and efficiency.

I also really wish that for gap filling Cura engine could treat the nozzle as whatever size needed. So make full width lines, until no more can fit. Then break up the rest into smaller than nominal beads. Because these gaps are bordered on all sides, I think you would achieve great gap filling, without needing to "judder" the nozzle like it currently does. It would also make huge improvements with axi-symmetric type parts, where the gaps are usually a consistent width and could be filled with one narrow bead. Currently, the engine struggles to recognize these reliably, and either over-extrudes, or fails to fill them at all.

BagelOrb, 2015-12-14:
I think we accounted for that; for example, we use a 0.35mm drill, so that the average nozzle diameter will come out at 0.4mm.

If you have a specific case where you find the gap filling algorithm is failing, please make a new github issue for that!
The new Cura should do better in filling these gaps than 15.04.

CCS86, 2015-12-14:
I think you guys haven't released a beta build since gap filling
improvements were made.

BagelOrb, 2015-12-14:
Gap filling had been improved months ago. We've made lots of builds. You
have tried 15.06 and/or 15.10 and/or other 2.0  pre releases right?

You can search the forum for the closed beta releases website..

If I remember correctly we do tests on windows 10 and all went well.

CCS86, 2015-12-14:
I am on the latest build, from 11/17, which is almost a month old now. I
know there was a double infill bug I reported that has been fixed since
then.

BagelOrb, 2015-12-14:
Could you please make a new github issue with a clear picture of where gap
filling goes wrong?

I know it's not always optimal, but at least most gaps should get filled
without overextrusion. Exceptions are perfectly diagonal gaps and gaps
smaller than 0.4mm in both length and width. Those might sometimes(!) not
get filling.