Ultimaker/CuraEngine #276: Feature Request: Volumetric Flow "Acceleration"

CCS86, 2015-11-21:
I believe there are big benefits to improving print speed without sacrificing print quality, by leveraging features like faster infill speed, wider infill lines, combined infill layers, slower outer loops, etc.

Unfortunately, in my experience, the more disparate the volumetric flow rate between regions, when using these settings, the more you end up with areas of over/under-extrusion in transition from one to another. Our bowden style extruder drives are naturally elastic in nature, and this leads to a transient response time measured in seconds, while the extruder pressure equalize towards steady state. 

My idea is to set a parameter which governs the maximum rate of change of volumetric flow rate, with respect to time. This would force the engine to break up the print moves immediately adjacent to a change in volumetric flow, with moves that step down or up to the new value. It seems like this even has the potential to mitigate the need for firmware based "pressure" advance.

If you evaluated the "next move flow rate" of a retracted rapid move as "0", it could even replace the need for the "coast" feature, because flow rate would already have stepped down smoothly towards zero; bleeding off extruder pressure without over/under-extrusion and allowing for far less retract distance.

What do you think?

BagelOrb, 2015-11-23:
Sounds like an intricate plan.

However, the bowden tube doesn't cause a delay in time, it's a loss in pressure. In order to accelerate the flow more pressure is needed. This extra pressure translates to extra filament being extruded. 

As long as the pressure advance is not implemented, an acceleration in flow results in underextrusion. By changing the flow gradually the underextrusion will only be more spread out - not negated.

Pressure advance could also be implemented in CuraEngine - at least: a version which advances based on the average flow within e.g. the infill.

This is currently out of scope.

CCS86, 2015-11-23:
I agree that it isn't a direct replacement for pressure advance, but still think it could help . You might be right though, that it just drags the under extrusion out farther.

Between the pressure lag, and the volume loss occurring during combing moves over empty portions of the layer, it becomes a real issue on some prints.