Ultimaker/CuraEngine #277: Struggling to fill gaps consistently

CCS86, 2015-11-21:
This model seems to be giving the engine fits. It is totally axisymmetric:

[Sheave.stl.pdf](https://github.com/Ultimaker/CuraEngine/files/40921/Sheave.stl.pdf)

This is with gap filling turned off:

[image]

And the same layer with it on, you can see that it is overlapping perimeters badly:

[image]

Another layer:

[image]

BagelOrb, 2015-11-23:
In the third picture I seem to see double infill, is that correct?

This might be quite a serious bug.

Filling inter perimeter gaps should leave the merged infill lines as they are.

BagelOrb, 2015-11-23:
Around layer 74 I see double infill.

BagelOrb, 2015-11-26:
Solved!

closed 2015-11-26 by BagelOrb

CCS86, 2015-11-26:
Sweet. Is this just in regards to the double infill, or also the sporadic disorder?

Ghostkeeper, 2015-11-27:
Just the double infill, as far as I can tell from the commits.

BagelOrb, 2015-11-27:
The sporadic disorder is not going to be fixed any time soon.