Ultimaker/CuraEngine #278: Toolhead 0 not set before starting FFF print

frazierjason, 2015-11-21:
When CuraEngine is slicing two enmeshed objects to be FFF printed together in different colors (toolheads) with a brim and a wipe/prime tower, the resulting Gcode fails to tell the printer which toolhead should be used when the print is started.  That is, there is no T[N] command at the start to define toolhead N to be used.  In my case, the firmware will end up using whatever toolhead was last selected, and if nothing was ever printed since firmware bootup, it picks the highest position toolhead (T1).

To ensure no assumptions about the toolhead in use, CuraEngine should always explicitly set the starting toolhead to be used, even if it is T0.

I'm using Repetier-Host 1.6.0 software on Windows 8.1 x64 with its integrated CuraEngine slicer.  The printer is a Cobblebot 1A running Repetier firmware 0.92.6 on a RAMPS 1.4 board.

Ghostkeeper, 2015-11-23:
Thanks for the report. I wonder if this is still a concern in version 15.04.03 or even the most recent version.

BagelOrb, 2015-11-23:
This is in fact still a problem.

Should be easy to fix.

markwal, 2015-11-26:
There's a couple of TODO in the sources: https://github.com/Ultimaker/CuraEngine/blob/master/src/FffGcodeWriter.cpp#L367
https://github.com/Ultimaker/CuraEngine/blob/master/src/FffGcodeWriter.h#L63

Seems like making the initializer in the header be -1 will always make the first set explicit, right?  Though perhaps an explicit setExtruder should be done before the first temperature set so that the temperature set without Tn parameter is on the correct nozzle?

The TODO's seem to indicate a desire for another setting as well though.

markwal, 2015-11-26:
I take it back -1 won't work because it's referenced as an array index before the first setExtruder.

BagelOrb, 2015-11-26:
Thanks for doing this prep work.

I was thinking about single extrusion machines. Could it be bad if they
also got a T0 command?

markwal, 2015-11-26:
I wonder.  Perhaps there are firmwares that barf when they see a Tn?  What you'd hope is that they ignore a T0.  But honestly, I'm not sure.

According to the sources, Marlin will accept a T0 even if it is compiled with EXTRUDERS 1.  But that is the only case I've looked at.

BagelOrb, 2015-11-26:
At http://reprap.org/wiki/G-code#T:_Select_Tool it says that every firmware accepts the command, except Sprinter and grbl.

I don't know if many printers use those firmwares, but it would be sensible to assume they handle unknown commands well (they skip them).

BagelOrb, 2015-12-16:
I have spent a great deal thinking about this and concluded that Cura must assume toolhead 0 to be selected at start. Otherwise it doesn't know on which extruder it is performing a retraction.

If your firmware doesn't ensure that toolhead 0 is selected you should manually insert the T0 command in the start gcode.

closed 2016-10-10 by BagelOrb