Ultimaker/CuraEngine #279: Zigzag Infill not connecting to perimeter

CCS86, 2015-11-23:
These two prints were with otherwise identical settings, one with lines infill and the other zigzag. The latter is not properly connecting infill to the perimeter:

[image]

[image]

Ghostkeeper, 2015-11-23:
Also, the improperly connected print seems to have 5 walls instead of 4.

BagelOrb, 2015-11-23:
ZigZag infill will make a line, and stop half the line width short of the inner walls and extrude during the move to the next line. 

```
Lines  ZigZag
|     |      |     |
|     |      |     |
| . . |      """"""
------------------ inner wall
------------------ outer wall
```

Generally choosing the top/bottom ZigZag pattern is a bad idea, but it's included because CuraEngine can.

Perhaps we should disable the option to keep users from trying it? Or do we want to give users the freedom to choose what they want?

CCS86, 2015-11-23:
Right, I understand how zigzag differs from lines, but both should work in my eyes. I don't think this lack of connection is acceptable for middle layers or top/bottom.

Basically, the "zigzag" portions should act like concentric infill, which should also connect. 

Maybe there just needs to be a static parameter applied to the half-nozzle width hold back, so that for the same infill overlap percentage, the prints look functionally the same between these infill styles?

Maybe it is the wipe distance alone that causes this difference. Using wipe with line infill can allow good connection without increasing the infill overlap (and causing over-extrusion in other areas, and is a super cool feature BTW). 

If zigzag doesn't currently benefit from this setting, maybe that wipe distance should be applied to the half-nozzle offset, to help normalize the different styles.

BagelOrb, 2015-11-24:
The gap between the top/bottom fill and the walls should be zero in the gcode; otherwise this would be a bug.

My guess is that the side piece next to the wall doesn't quite catch on to the wall and gets dragged along with the next infill line backward.

Here's a picture of why zigzag infill is a bad idea for top/bottom. The orange pieces are where there is likely to be no filament, cause the corners get rounded.
[image]

If you apply the on offset of half the nozzle width to the zigzag portions but extrude the same amount you end up with lines infill. I think you should just stay with lines infill for top/bottom skin.

CCS86, 2015-11-24:
Are you saying that bottom/top layers handle zigzag infill differently than other layers? Do they not use infill overlap?

BagelOrb, 2015-11-24:
Skin doesn't use the infill overlap, only the (sparse) infill uses the infill overlap.

For sparse infill it's not so bad to have a little overextrusion, since it is sparse and there's enough place to go for the overextruded filament.

For skin its worse to have an overlap with the walls.

CCS86, 2015-11-24:
I'd argue that a touch of over extrusion on dense infill, skin or not, is
far better than a severe discontinuity.

I print a fair amount of dense parts and have no problem tuning extrusion,
with the infill overlap in effect

BagelOrb, 2015-11-24:
When you set infill density to 100%, all infill is replaced by skin, so you won't see any overlap.

You reasoning is sound, though. Overextrusion is better than a severe discontinuity.

However, why not just print with lines top/bottom infill? Then you won't have either problem!
Why are you pressing on the zigzag infill so much?

CCS86, 2015-11-24:
I am not opposed to using lines infill, but just think that zigzag should
work as well, if the user so chooses. I'd hate to see it axed as a feature,
when all it needs is a small tweak.

BagelOrb, 2015-11-25:
We are currently quite swamped, but you are free to submit a PR.

CCS86, 2015-11-25:
I hear ya, and for the record, the latest beta build is pretty great! You guys have been steadily improving it, and it feels like it is getting pretty close. :)

Is there a reason not to simply apply infill overlap in these cases? Or if that really proves to be too much material, a fixed scalar * overlap? I wouldn't imagine the code change would be too complex for that.

BagelOrb, 2015-11-25:
We could introduce a new overlap: the skin overlap, however, the infill overlap already was a hack to make the infill work, even though overextrusion would take place.

I might give it a go sometime. Shouldn't be too hard to implement.

CCS86, 2015-11-25:
Glad to hear it :+1: 

I know that infill overlap seems like a hack, and maybe it is a ridiculously simplified "mathematical model" of what is really taking place... but for how simple it is, it works very well! 

I see it as a compensation for a number of physical things that impede connection, including: 

-print head acceleration : extruder pressure mismatch
-XY motion system lash/stretch
-filament surface tension and inertial effects
-perimeter and infill over/under-extrusion

Having a single linear scalar that combats the combined effects of all these things with a decent level of success is pretty cool. Every printer is going to be different in the above listed areas and more, so letting the user fine tune overlap for their rig saves you guys a ton of complex modeling.

I never realized that when setting the infill to 100%, you effectively disable the infill overlap parameter ( a testament to how well the line infill plus wipe distance works!). Maybe instead of having a "skin infill overlap" parameter, you could leverage the existing infill wipe distance:

Use the infil wipe distance to extend the position of the zig (or zag? :) ) move closer to the perimeter, but do not add to the extrusion volume. This should give a similar effect where the exaggerated print head movement helps connect, without commanding an extrusion volume that causes over-extrusion.

BagelOrb, 2016-03-09:
I've implemented infill overlap and infill wipe distance to work on skin as well for the 2.1 release. In the 2.2 these settings will be split into separate settings for infill and skin.
Infill wipe distance only works for linear types of infill pattern (lines, grid, triangles).

Your idea about extending zigs without altering volume is quite a hard one to implement, with little gain, so I don't think we will do that.

closed 2016-03-09 by BagelOrb