Ultimaker/CuraEngine #280: Coast vs Retract

BagelOrb, 2015-11-26:
Retraction coasting currently retracts before it coasts and thereby also performs the zhop. 

Retraction should occur after coasting.

CCS86, 2015-11-30:
Is it possible to evaluate an additional coast parameter "overshoot", while it is being looked at?

I've done a bit more testing on coast today, and found a few things. 

If you try to reduce speed by more than the default 10%, there is an immediate pressure to required flow discrepancy, which causes a bulge of over-extrusion. (A coast velocity ramp being the best simple solution I see)

If you increase coast volume to an amount that successfully bleeds off most of the extruder pressure, you end up with a small divot at the seem, for the same reason above. If the coast move slightly overshot the seam, similar to infill "wipe", I think it would help smooth out seams.

BagelOrb, 2015-12-01:
I would gladly implement these ideas, but currently I'm swamped.

I think you have the right ideas. Did you test the 10% against e.g. 5% ? I suspect that the bulge you see already comes from the fact that the retraction is performed before coasting, which causes the nozzle to leak during the retraction. Moreover, a 10% flow decrease should never result in over-extrusion.

BagelOrb, 2015-12-16:
See https://github.com/Ultimaker/CuraEngine/issues/288 for your suggestions.

closed 2015-12-16 by BagelOrb