Ultimaker/CuraEngine #282: follow mesh / surface mode surface ignores retraction

rooiejoris, 2015-12-01:
For a special project I need more control of the print path and use the very nice follow mesh [new cura: surface mode surface] option. But..... retraction is somehow ignored.

in the cura engine log [see below] you can see that retrection is on

[side question, how to see the 'cura engine log in the 'new' cura...?!]

cheers / joris

Ghostkeeper, 2015-12-02:
As for your side question, it's not really possible any more since after every slice or crash the engine is restarted, which overwrites the engine log file. That's currently a bug too. It's related to an internal issue that the engine is nondeterministic; when you slice multiple identical objects with it without restarting, the outcome may be different. To fix that we had to restart the engine every time. Sorry! We'll figure that one out some time.

BagelOrb, 2015-12-02:
Perhaps the retraction limitation is kicking in here.

Make sure combing is on and set the Minimum extrusion distance window to 0.0, in order to never limit the amount of retractions occurring over the same patch of filament.

rooiejoris, 2015-12-03:
also tried with all retraction setting in zero, but doesn't seem to retract either [didn't print, but checked the gcode]