Ultimaker/CuraEngine #283: object sink

rooiejoris, 2015-12-05:
from 15.04_to_2.0.txt:
objectSink              [ Doesn't exist in CuraEngine anymore ]

why doesn't this exist anymore?
For me this is a very usefull setting to 'flatten' not perfect stl files, but also to test some settings on a differcult part further up. I also know that people use it to resume a [failed] print form a specific height.

Even better, It would be perfect to also have a "cuttop" option so you really can test print some difficult areas in the object.

[workaround now is meshmixer, but it is quite annoying for small changes in the model to open and modify the file in a separate program]

cheers / joris

Ghostkeeper, 2015-12-06:
If you move the object below the build platform in the front-end, it cuts the object off too. This however is not a solution for people who use a different front-end such as the command line or a script, like you, and doesn't allow for very precise cut-offs. So I'll mark this as an enhancement and maybe it'll be implemented again.

No clue why it was removed from the engine.

BagelOrb, 2015-12-06:
The new CuraEngine has mesh_position_z with which you can do exactly that...

rooiejoris, 2015-12-06:
Ah ok.
I can add this in the fdmprinter.json somehow no!? Any hints?

(And the txt file can be updated then.)

Cheers / Joris

BagelOrb, 2015-12-06:
Fdmprinter.json is the file for letting the frontend know which settings
there are. However, the object position is directly modified in the
frontend, so the settings shouldn't be in fdmprinter.json.

You should pass the setting as command line option.
Putting it as a new setting in the json is a dirty hack, but it will work...

rooiejoris, 2015-12-06:
>> If you move the object below the build platform in the front-end
how can I do that...?!

ah, figured it out, this is not possible after a scale, i think i read something like this in asana...

nallath, 2015-12-07:
It should be possible after a scale.

rooiejoris, 2015-12-08:
>> Putting it as a new setting in the json is a dirty hack, but it will work...

This works perfect [don't understand exactly why this is a dirty hack], also tested it with moving the object below the platform. Works as well.

Also interesting for me are mesh_position_y and mesh_position_x, will test that also one time...

[another feature could be "adopt position from stl" so you can easily print on top of somthing... or is this also hackable?]]

cheers / joris

BagelOrb, 2015-12-08:
It's unclear to me whether you want to use CuraEngine as a command line program or you are using the frontend.

The json is principally for use with the frontend, which means that the mesh_position_z shouldn't be in there. The mesh_position_x y and z are to be used when using CuraEngine as standalone program.

If you're using the frontend, just move the object with the mouse.

If you want to have two objects above eachother, you should have them positioned above each other in the STLs and then in the frontend select them, right click, select Merge objects. Then the relative positions between the two STLs is maintained.

closed 2015-12-08 by BagelOrb

nallath, 2015-12-08:
Mesh_position won't work as a setting when using multiple meshes either. That also makes it a hack.

BagelOrb, 2015-12-08:
It should work when using CuraEngine as command line tool alone without the frontend....
Doesn't it?

nallath, 2015-12-08:
Only if you want all of the objects to be sunken.

BagelOrb, 2015-12-08:
Ah yes, that might be a bug.

BagelOrb, 2015-12-08:
Are you sure? Looking at the code it should work per object...