Ultimaker/CuraEngine #285: Mid-air Support

CCS86, 2015-12-15:
Hi guys,

Is there a setting that can help control this support being generated in mid air? I have not had much luck yet.

[image]

BagelOrb, 2015-12-15:
You could set the support bottom stair step height to zero.

BagelOrb, 2015-12-16:
If you select support pattern grid it will not generate such mid air blips.

BagelOrb, 2015-12-16:
In your picture these mid air pieces are generated by the support roofs.

You could also try the Concentric support roof pattern.

I have no idea how to solve this problem. I think it would happen quite often when using zigzag support with lines roof.

CCS86, 2015-12-22:
It seems to be due to the fact that the inside radius of the part causes Cura to generate roof closest to the model first. Because of the large XY distance I specified, I did not expect this to happen. It's a 1mm radius, with an XY separation of 1.5mm.

Maybe there is a small error in how that calculation is done?

BagelOrb, 2016-01-04:
If the support looks like this
```
 __
/  \
|  | (side view)
```

Then the horizontal part as well as the diagonal parts will get support roof. The diagonal parts also have a horizontal aspect.

The way support is generated is as follows: first the volume in which to generate the support structure is calculated and then the roof is generated as if it was top skin.

I now see that this can be improved upon, since some diagonal parts are introduced by the x/y separation.

BagelOrb, 2016-10-10:
Fixed

closed 2016-10-10 by BagelOrb