Ultimaker/CuraEngine #286: Trouble with gap filling

CCS86, 2015-12-15:
Hi guys,

I'm seeing some issues with gap filling. I have it turned on everywhere in this gcode, but it is still struggling:

[image]

More trouble:
[image]

I'm not sure if this is a display issue in repetier, or what, but this is with coasting turned off to see if it played a part:
[image]

Yikes!
[image]

CCS86, 2015-12-15:
Gcode:

[UMO_Idler Carrier_nocoast.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/61918/UMO_Idler.Carrier_nocoast.gcode.txt)

Model:

CCS86, 2015-12-15:
Won't accept the zip for some reason. Just drop the .txt extension and unzip:

[Idler Carrier.zip.txt](https://github.com/Ultimaker/CuraEngine/files/61919/Idler.Carrier.zip.txt)

BagelOrb, 2015-12-15:
The bug you are seeing in the first two images is a bug of mergeInfillLines. When CuraEngine sees consecutive small line segments `iiiii` it will convert them into one line segment in the orthogonal direction `====`.

The bug is that it apparently also converts non-consecutive small line segments `i  i  i` into a line segment `-------`.

This bug can be seen in the first picture: on the left side they are combined and on the right side they are not.
On the right upper side of the second picture they even seem to have been combined in a quite horrible fashion.

I also think I'm seeing both skin and infill on 100%. I recommend making the top or bottom skin practically infinite, so that all infill is replaced by skin. That wat you don't get the separate areas on the other sides of the walls where the pointy things are.

The third picture seems to be of a different layer - where the pointy things are connected; otherwise it might be a Repetier visualization bug. Try viewing that layer in Cura 15.04 to see if the pointy things really are connected.

BagelOrb, 2015-12-15:
I'm afraid this will not be solved in the coming release, but the one after..

CCS86, 2015-12-15:
You are correct, the 3rd picture is a different layer. These are some printed threads, with the screw axis on the XY plane, if that helps explain it. 

I should have drawn arrows there too, but I posted it because the gap filling looked bizarre.

CCS86, 2015-12-15:
Is it possible to have a flag when 100% infill is used, that it should skip the algorithm which tries to section off infill regions which are near overhangs, and fill them 100%.

BagelOrb, 2015-12-15:
If you select 100% infill in the frontend, the infill is automatically
turned off, bottom layers set to 9999 and top layers set to 0...
This is done in the frontend, not in CuraEngine.

CCS86, 2015-12-15:
So, why is it that areas which will soon become overhangs, are still sectioned off and filled separately. I thought that feature was meant to help densify sparse infill to better support overhung areas.

CCS86, 2015-12-16:
I am not seeing the behavior you mentioned:

"If you select 100% infill in the frontend, the infill is automatically
turned off, bottom layers set to 9999 and top layers set to 0..."

With infill set to 100% "lines" and bottom layers 5, the 12th layer looks like this:

[image]

Changing bottom layers to 100, and it looks like this:

[image]

[UMO_Sheave i-nocoast-5-bot.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/63432/UMO_Sheave.i-nocoast-5-bot.gcode.txt)
[UMO_Sheave i-nocoast-100-bot.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/63433/UMO_Sheave.i-nocoast-100-bot.gcode.txt)

Model:
[Sheave i.stl.txt](https://github.com/Ultimaker/CuraEngine/files/63434/Sheave.i.stl.txt)

CCS86, 2015-12-16:
It gets soooo close on some of these layers, then just falls apart in a couple spots!

[image]

BagelOrb, 2015-12-16:
Perhaps this has been fixed in between your version and mine, cause I am seeing the 100% infill thing working.

[image]

Your last picture is another problem of Merge infill lines again :(

BagelOrb, 2015-12-16:
Ah no your last picture is not a mirge infill lines issue per se. The thing is that the infill is converted to a smooth round line everywhere, except for the places where the infill is approximately aligned with the direction of the circle. It's a limitation of how that small area is converted into two long circular lines.
That will not be fixed any time soon. :/

CCS86, 2015-12-17:
Shoot, that's disappointing. Is that feature useful in other situations?

BagelOrb, 2015-12-18:
I fixed this issue in the current release accidentally, so perhaps it will be in the next release after all, instead of the one after that.

closed 2015-12-18 by BagelOrb

BagelOrb, 2015-12-18:
Even with circular infill areas, the revised algorithm works better! :D

CCS86, 2015-12-18:
Excellent, nice work my friend!

Is there anyone we can tag here who might be willing to compile a new beta build? I've got some new prototype parts to print out and would love to test some of the fixes.

Ghostkeeper, 2015-12-21:
I can send you a build for Linux if you like, or Windows 32-bit with some work, but other ones have to wait. @awhiemstra is the only one able to make a proper OSX build here and he's on vacation.

Our weekly builds have moved internal since our testers need a build that includes secret information, so we can't share those, but I can make a build without all that for Linux or Win32 (not 100% sure about Win32 but it should work).

CCS86, 2015-12-21:
That would be great. I've been using Win64, but 32 should be fine.

Ghostkeeper, 2015-12-21:
I'll give it a try for you once I'm home tonight. No promises though! :stuck_out_tongue: I'll keep you updated. The Win64 build has some problems with building Python libraries which are out of my league.

Ghostkeeper, 2015-12-22:
I successfully built it for win32 but running it caused a complaint that it couldn't find Qt libraries, and it became fairly late (2:00). Going to try again tonight.

CCS86, 2015-12-22:
I'll sign an NDA if that makes things easier ;)

Ghostkeeper, 2015-12-23:
It would, but I'm not at liberty to make the decision to offer that to you :wink: I'd only distribute public builds here. No luck so far though, still getting a runtime error after running the installed binary: Ultimaker/cura-build#26 . A senior developer is working with daily internal builds right now too, so if he's having success I could ask him.

To keep this issue thread on topic, I'll try to keep radio silence from now on unless I can offer a (non-secret) build and you'd like to help test the fix.