Ultimaker/CuraEngine #287: Missing Support Lines

CCS86, 2015-12-15:
I'm seeing two issues with support generation. One is that some layers are missing portions of the intended support line:

Complete:
[image]

Incomplete:
[image]
[image]

Also, if you look at the progression. The support in this region is stair stepping outwards, but is doing so in fairly large increments. I'm not sure this would physically work.

Both the gcode file and the model can be found here:

https://github.com/Ultimaker/CuraEngine/issues/286

Ghostkeeper, 2015-12-16:
Thanks for the report. Seems that at the least the default step size needs to be tweaked, and at the most there is still a bug with the support areas.

BagelOrb, 2015-12-16:
As for the missing support lines: see https://github.com/Ultimaker/CuraEngine/issues/265 .

The bottom stair step height will create steps which bridge gaps depending on the angle of the model where it sits. If the bottom has a very slight angle, the steps will be quite big. However, the stair step height is quite low by default (0.3mm), so the lines don't really have the chance to sag.

Note that a smaller X/Y distance might also help to make those new steps sit on the model without falling too much.

CCS86, 2015-12-16:
I'm a little confused, #265 was marked closed because this issue (#287) is a better description. But this issue is marked closed as well.

BagelOrb, 2015-12-16:
No, this issue is marked as a bug ;)

CCS86, 2015-12-16:
My bad :)

BagelOrb, 2016-01-25:
[ SOLVED ]
:D

closed 2016-01-25 by BagelOrb