Ultimaker/CuraEngine #288: Polygon Overshoot

BagelOrb, 2015-12-16:
Perform an travel move called 'overshoot' after printing each polygon, so that the start and end of a polygon attach better.

After printing a polygon, make a small move in the direction of the first line segment.

CCS86, 2015-12-16:
Sweet!

CCS86, 2015-12-16:
Would this be specific to coast moves?

BagelOrb, 2015-12-16:
I don't see why.

Actually, now that I think of it, it doesn't really make sense to have this feature. It already ends exactly where it began, so there already is a complete overlap between the nozzle hole area and the already printed start line segment.

What was your idea for requesting this?

CCS86, 2015-12-16:
Just to minimize seams. Regardless of coast volume, I see a pucker inwards at the seam. By keeping the coast speed at 100%, it avoids bulging out, but I get a small pucker inwards with coast volume very small ( ~.02 mm^3 ), or large (~.23 mm^3).

CCS86, 2015-12-16:
I am really anxious to get a new beta build with the "coast retract order" bug fixed, so I can keep testing it.

I'm up to .23 mm^3 move-coast volume, and the seams are getting really good. I haven't tried longer coast moves yet.

BagelOrb, 2015-12-17:
I think the pucker inward is the downside to not having the bulging out. I
think that the problem should be resolved by gradually turning down the
speed, as you mentioned in another issue.
Now, at the end of a coast the pressure is lowest and so the maximal flow
is probably too low for the bleeding off to ooze the correct amount.

BagelOrb, 2017-01-06:
Implemented

closed 2017-01-06 by BagelOrb