Ultimaker/CuraEngine #289: Fast Feed Moves?

CCS86, 2015-12-21:
Hi guys,

I just did some poking around in a generated gcode file, and found a bunch of commanded feeds that are far faster than anything I commanded:

[image]

[image]

Any ideas what's going on here?

Gcode:
[UMO_B41033019.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/68928/UMO_B41033019.gcode.txt)

CCS86, 2015-12-21:
I went through and parsed all the feed rates from this gcode file, and attempted to clear out the duplicate values to see all the unique values, but eventually got tired and stopped. There are a crazy amount of different feed rates posted. I don't get it!

[image]

CCS86, 2015-12-21:
The only ones that make sense to me are:

2100 (print speed)
4200 (travel speed)
1440 (retract)
1080 (prime)
900 (bottom layer)

Ghostkeeper, 2015-12-22:
Cura can adapt the speed to get a different line width. I think in this case it is probably merging infill lines, or else filling small gaps.

CCS86, 2015-12-22:
Is there a specific reason that a multiplier is applied to speed instead of extrusion volume/rate? This seems like bad practice. IMO, the gcode should never try to move the machine faster than you've specified, even over short distances.

Other issues with that strategy would be that by increasing speed, you increase the percentage of the move which is either accel or decel, which gives you even less control over relative extrusion rate.

Ghostkeeper, 2015-12-23:
Good question. I don't know, honestly. Maybe it is easier to control due to oozing? It seems that especially for filling small gaps it'll be accelerating throughout the entire move.

The people that implemented this are on vacation, so I think it'll be some time before this gets an answer.

BagelOrb, 2016-01-04:
This was implemented as to keep the flow (extruded volume per second) more constant. The speed is capped at 9000, which is a hardcoded value at the moment. 

Have you experienced specific problems with this strategy?

Now that you know the reasons behind the implementation, what strategy do you think would result in better quality?

CCS86, 2016-01-04:
I can't say that I have specific problems occurring from this. I only
noticed it when looking through some gcode I generated to try on a very
large format printer (~1 m^3)

I can see what you are going for in this. Basically, you are banking on
being able to change print head velocity, faster than you can extruder
pressure. I haven't done the math on it, but seems believable to me. It's
clever.

I think problems would start arising when the machine's Vmax was capped
below the commanded velocities. You would lose control of total extruded
volume. This would also happen, when low machine acceleration settings
cause a significant portion of the move to be done during print head
acceleration. I think the result would always be over extrusion.

Have you done a comparison test for quality between varying head speed vs
varying extrusion? Maybe a combination of both could help minimize error.

BagelOrb, 2016-01-04:
I haven't done any empirical testing. Such tests can easily become quite laborious.

I should perform these tests some time in the future, but currently it is not that high on my TODO list.

BagelOrb, 2016-10-10:
The issue is also solved now. You can toggle whether you want to apply speed changes for smaller line segment.

closed 2016-10-10 by BagelOrb