Ultimaker/CuraEngine #291: Small slicing issue with version 15.04.3

sanchosk, 2016-01-04:
I have Cura version 15.04.3 installed.
I noticed something strange.
I downloaded this thing:
http://www.thingiverse.com/thing:707815/#files
When slicing the object Drawers_case_no_top.stl, everything is OK.
But, as I had a space on the printbed, I put also 1bin_drawer.stl next to it.
I printed it and noticed something strange - between layers 9 to 11 (I tried it multiple times) the slicer adds a layer with full fill on the case part. This does not happen when the part is sliced separately.
I will try to attach the Gcode, but I'm not sure the size will not be an issue here.

[drawers_error.zip](https://github.com/Ultimaker/CuraEngine/files/77575/drawers_error.zip)

Ghostkeeper, 2016-01-05:
Have you tried this also with version 2.1? It can be downloaded here: http://software.ultimaker.com/Cura_closed_beta/15.10/

This repository concerns issues from version 15.06 onward, since much of the program was rewritten back then and ownership changed. Issues regarding version 15.04.3 and before should be put in the old repository: https://github.com/daid/Cura/

There is a good chance that your bug has already been fixed in the newest version!

BagelOrb, 2016-01-06:
Actually the slicing process (creating the outlines of each layer) is one of the aspects which hasn't changed at all since 15.04. 

I cannot reproduce the problem.

You can try fiddling with the Fix Horrible settings in the expert settings menu and see whether you can solve the problem. First try turning off all fir horrible options.

closed 2016-10-10 by BagelOrb