Ultimaker/CuraEngine #292: Drawing perimeters from the outside in

frank16, 2016-01-06:
I noticed that when I print a thick perimeter, say 1.5mm, it is printed from the inside out (the head goes around a few times). This makes the outside layer bulge a little messing with dimensional accuracy. Would it be possible to have an option to print from the outside in?

BagelOrb, 2016-01-06:
Indeed the inner walls are printed first, before the outer wall. 
This helps with overhangs, as then the inner walls give the outer wall something to hold on to.

The outer wall is in several aspects differently configurable from the other walls (line width for example).
There are several possibilities regarding the ordering of walls begin printed:

1. innermost wall -> second wall -> outer wall (current)
2. second wall -> innermost wall -> outer wall (inner walls reversed)
3. outer wall -> innermost wall -> second wall (outer wall first)
4. outer wall -> second wall -> innermost wall (total reversed)

It would be a bit too much to implement all these possibilities.

Which one would you think would suit your situation best?

frank16, 2016-01-06:
How about outer wall first unless it's an overhang? If it's an overhang 
exact size is probably less critical. So situation 4 by default, 1 if 
overhang, that would seem to do it? But then what do you do when you go 
from normal outerwall to overhang, is your next question, right ;-)
But wait a second, the real issue is the bulge on the outer wall, right? 
That kills precision. But it is fixable by lowering extrusion rate 
slightly, you could even calculate the exact amount I guess, based on a 
hardened second ring. But then you would draw that second ring first, then 
calculate extrusion based on hardened second and half open outer. Auch, 
this is getting mathematical quickly :-(
The nice thing would be that for holes this might work very well too. The 
second wall defines the size and sticks to the layer underneath, the inner 
(bordering air) sticks to it in turn and is less 'tempted' to take a 
shortcut. The slight underextrusion helps further in preventing hole shrink.

I hope we mean the same thing with 'inner' (bordering infill) 'second' 
(fully filled) and 'outer' (bordering air)? (that's how I understood it)
Thanks for the challenge :-)

BagelOrb, 2016-03-07:
Sorry for the late reply!

It's not possible to let the order depend on whether 'it' is an overhang, since overhang areas can cover part of the walls. Therefore I think option 1 is the safest.

I don't really like the idea of lowering extrusion, but it shouldn't be that hard to implement.
There already is an extrusion rate which is passed to every feature (walls, infill etc).
It shouldn't be very hard to configure that extrusion rate per feature type separately...

(Yes we mean the same thing ;)

CCS86, 2016-03-07:
In my eyes, if printing the outer perimeter last is causing bulging and
dimensional inaccuracy, then you don't have the print well tuned, and are
over extruding.

BagelOrb, 2016-10-10:
There's anow an option Outer Before Inner Walls.

closed 2016-10-10 by BagelOrb