Ultimaker/CuraEngine #294: Does coasting before retraction travel work?

Dim3nsioneer, 2016-01-21:
I tried the coasting feature and got nice results for non-retraction travel moves but couldn't get a slice where coasting before a travel move with retraction happened. Am I doing something the wrong way or is it simply not yet working?

CCS86, 2016-01-21:
It was "working" on one build, but I found the order was wrong. It was
stopping where the coast move begins, retracting, then coasting.

Now it just seems to be broken.

BagelOrb, 2016-01-22:
What do you mean it just seems broken?

What's wrong?

CCS86, 2016-01-22:
This is with the 11/18 build  so I'm not sure how relevant it is. With
certain values for the coast volume it wasn't slicing

BagelOrb, 2016-01-22:
I haven't worked on coasting for a while. What kind of values for which settings? Making the coasting volume really high? Something like that?

Dim3nsioneer, 2016-01-22:
@BagelOrb In my case I didn't get Cura to replace any print line with a travel line before a retract (UM2 profile). It worked without retract.

CCS86, 2016-01-23:
Nothing too crazy on values. Maybe 0.2mm^3 coast volume, and 2mm^3 volume before coast.

BagelOrb, 2016-01-25:
I am seeing coasting with those setting values - both before a retraction an before a normal move. Perhaps it has been fixed already then.

The last public release of Cura 2 was already quite some time ago.
Hopefully we can release soon, so you can really benefit from coasting!

closed 2016-01-25 by BagelOrb

Dim3nsioneer, 2016-01-25:
Well, what about bringing another beta first? It's not a bad thing IMHO if the very last beta and the released version are almost identical...

CCS86, 2016-01-27:
That would be great!

I am unable to test build 2.099-2.1 because of:
https://github.com/Ultimaker/Cura/issues/594

BagelOrb, 2016-01-27:
I think the idea is to first do a closed beta release for (semi-)internal testing, then an open beta and after that the official release. 

I call a beta a "release" as well...