Ultimaker/CuraEngine #295: skirt should be a convex hull of all parts, not surrounding each part

hariedo, 2016-01-23:
Currently (15.04.4), the skirt is an inflated perimeter of each part on the bed. If the part is concave, the skirt wanders into the concave areas. If the part has large holes, additional skirt lines are drawn inside the part. Both of these behaviors cause the skirt to take longer, and to potentially interfere with the nozzle as it moves around between parts (since the skirt is the first extrudate and often stringy or blobby or loose).

The purpose of the skirt is not to mimic the part but to (1) allow extrusion to rise to expected smoothness and adhesion before starting the parts, and (2) allow a general check of bed tramming ("leveling") across the print area.  Extra details in the skirt that venture into the middle of the print area can be problematic.

I propose the path of the skirt be simplified to the convex hull of ALL of the current skirt's paths. The convex hull is the minimum convex shape that fits all the points inside. Once the convex skirt is drawn, all future layer-1 movements occur inside this perimeter so the skirt plastic cannot interfere.

This is actually the behavior seen in Slic3r for years now, and it is a functional choice, not an aesthetic choice.

BagelOrb, 2016-01-25:
The new Cura had the skirt mimicking a convex hull. Couldn't be bothered to
actually implement it, since the approximation took me only 5 minutes.

The new cure will be released soon.

Ghostkeeper, 2016-01-25:
Newer versions of the engine compute the skin by giving the model a very large offset (the skirt width plus 10cm), then reducing that offset by 10cm again. It ends up at the convex hull for small objects, and for objects that are very far apart it will not try to link them together in one convex hull. I think it works nice.

daid, 2016-01-25:
Note that it was not convex for 2 reasons. One was that I didn't have a drop-in algorithm at the time I made the skirt. Second is because the skirt and the brim shared implementation, and the brim should not be a convex polygon.

closed 2016-10-10 by BagelOrb