Ultimaker/CuraEngine #296: How to change the setting to disable filling the gaps?

ynilsiam, 2016-01-28:
Hi,

I set the infill pattern to concentric. However, when it get to the middle of the mesh, it went crazy try to fill the gap in the middle.
What is the setting parameter to disable filling the gap feature?
Is there any practical way to avoid having the gap?

Thank you very much!

Ghostkeeper, 2016-01-29:
The concentric infill will always have some very small moves in the centre, especially if your item is round or approaching roundness. There is no way to disable filling the gap in the centre. If you have a gap with a diameter less than one line width, then you can try to fill it by slightly adjusting the infill density, e.g. from 94 to 94.1.

There is a setting in the Shell category that's called "Fill Gaps Between Walls", which is by default set to "Everywhere". You can disable that, but I don't think that's the one you mean.

BagelOrb, 2016-01-29:
The best way would be to choose line infill instead of concentric.

closed 2016-03-07 by BagelOrb