Ultimaker/CuraEngine #301: build fails, arcus/error.h no such file or directory

Robertbaelde, 2016-02-09:
I get the following error when i try to build the cura engine. 
I've installed all dependency's. 
``
[  2%] Built target clipper
[  4%] Building CXX object CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o
/root/cura/CuraEngine/src/commandSocket.cpp:12:25: fatal error: Arcus/Error.h: No such file or directory
 #include <Arcus/Error.h>
                         ^
compilation terminated.
make[2]: *** [CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o] Error 1
make[1]: *** [CMakeFiles/_CuraEngine.dir/all] Error 2
make: *** [all] Error 2
``

ynilsiam, 2016-02-09:
You need to Install libArcus (see https://github.com/Ultimaker/libArcus).

Robertbaelde, 2016-02-09:
i've already installed that using the instructions provided..

nallath, 2016-02-10:
Did you install the right branch? We've been messing around with how libArcus works.

Robertbaelde, 2016-02-10:
I installed the master branch, should i use another one?

Ghostkeeper, 2016-02-10:
We should really backmerge those changes, by the way...

You seem to have CuraEngine in the 2.1 branch, Robert. There's a 2.1 branch for libArcus as well.

nallath, 2016-02-10:
Or use CuraEngine in master and LibArcus in 2.1

pietrondo, 2016-02-10:
Same problem with archlinux and cura-engine-git

pietrondo, 2016-02-10:
```
src/CuraEngine/src/commandSocket.cpp:12:25: fatal error: Arcus/Error.h: No such file or directory
compilation terminated.
CMakeFiles/_CuraEngine.dir/build.make:118: recipe for target 'CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o' failed
make[2]: *** [CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o] Error 1
CMakeFiles/Makefile2:105: recipe for target 'CMakeFiles/_CuraEngine.dir/all' failed
make[1]: *** [CMakeFiles/_CuraEngine.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

BagelOrb, 2016-02-11:
Did you try building libArcus from the 2.1 branch instead of master?

Robertbaelde, 2016-02-13:
Tried installing it with the libArcus 2.1 branch and it worked!

closed 2016-03-07 by BagelOrb

awhiemstra, 2016-03-08:
Everything should now also be in master, so that should be ok now.