Ultimaker/CuraEngine #302: There should be a retract after parts finish, and there shouldn't be a raise Z

mikenew12, 2016-02-11:
When a part finishes the print head slowly rises a few mm and leaves a little deposit of filament on top of the finished part. This happens for a single part and when moving from one part to another for one-at-a-time printing. 

[image]

I took this piece of gcode (this is right when one part finishes and it's moving to the next one):
```
M107
G92 E0
G0 X166.962 Y29.629 Z43.194
G0 X43.057 Y63.525
;LAYER_COUNT:191
;LAYER:0
G0 X43.057 Y63.525 Z0.200
G0 X46.122 Y66.418
G0 X47.045 Y65.547
;TYPE:SKIRT
G1 F900.000 X47.992 Y64.740 E0.14931
```
and changed it to this:
```
M107
G92 E0
G10 ;       <--- add retract right as the first part finishes
;           <--- no more raise Z command
G0 X43.057 Y63.525
;LAYER_COUNT:191
;LAYER:0
G0 X43.057 Y63.525 Z0.200
G0 X46.122 Y66.418
G0 X47.045 Y65.547
;TYPE:SKIRT
G11 ;       <--- un-retract right before new part starts printing
G1 F900.000 X47.992 Y64.740 E0.14931
```
Now the parts print perfectly with no little stalagmite on top, and I can do one-at-a-time printing without needing to print skirts since the extruder hasn't leaked in between (I know mileage may vary depending on how leaky the extruder is, but for me printing PLA it works perfectly).

So basically, my suggestion is:
1) There should be a retract as soon as a part finishes.
2) There's no need for a raise Z command when a part is done. I'd rather have my own "end g-code" take care of moving the extruder out of the way when a print is finished since I may want to do certain things with it like parking it in a particular location.

Ghostkeeper, 2016-02-11:
I thought we fixed this last December. What version of Cura (or CuraEngine) are you using?

BagelOrb, 2016-02-11:
Well, he probably didn't build Cura himself, so it's probably the previous release.

This issue has been fixed in the next release.

closed 2016-02-11 by BagelOrb

mikenew12, 2016-02-11:
You're right, this was the last beta build from November. 

I did actually build master a couple weeks ago and try it there, but one-at-a-time was completely broken as the print head would just crash into the existing part. So I wasn't able to test this. But anyway, great to hear it's fixed.