Ultimaker/CuraEngine #305: Wierd raster through model using default machine templates

Robertbaelde, 2016-02-13:
Hi,

When i run the engine to slice a simpel test model, i get a wierd raster through the model.
This raster starts at layer 6 and runs arround/through the entire model. 

I used the standard cura machine settings for a prusa i3. 

Is this normal? How can i fix this? See sreenshots from the gcode (cura) for example

[image]
[image]

Gcode (remove .txt) 
[blokje.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/129434/blokje.gcode.txt)

the stl file
[blokje_10_20_30.zip](https://github.com/Ultimaker/CuraEngine/files/129435/blokje_10_20_30.zip)

I've tested it with other stl files, and those have the same issue..

BagelOrb, 2016-02-14:
Your fdmprinter.json is out of date. Probably your Cura repo is still on
the 2.1 branch while CuraEngine is on the master branch..

What you are seeing is an infill overlap of 10mm instead of 10%

BagelOrb, 2016-02-14:
Current development is on a 2.1 branch for each repo involved.

Robertbaelde, 2016-02-14:
tnx, switching to 2.1 worked

closed 2016-02-14 by Robertbaelde