Ultimaker/CuraEngine #308: Inconsistent coasting distance

CCS86, 2016-02-23:
Hi guys,

I'm running the 2/20 build and am seeing what appear to be different coasting distances/ volumes between outer and inner perimeters. 

What do you think?

[image]

BagelOrb, 2016-02-23:
It seems to me that the inner perimeters don't have enough volume to build
up the pressure to coast the full amount.

This is a feature, not a bug ;)

closed 2016-02-23 by Ghostkeeper

CCS86, 2016-02-23:
I should have read the new tool tip!

I didn't realized you had implemented linear scaling to the coasted volume. Awesome!

So, now there is no separation of comb/retracted coast moves?

BagelOrb, 2016-02-23:
nope, both retraction coasting and non-retraction coasting are the same... The retraction is delayed until after the coast move.