Ultimaker/CuraEngine #310: Feature request help: Bypass STL slicing with presliced geometry (includes topic branch)

codehero, 2016-02-25:
In my application, I am easily able to generate cross sections of my data. Converting to STL to be sliced again is inefficient and potentially lossy. I have hacked in initial support of reading the slicer data directly.

[https://github.com/analog10/CuraEngine/tree/direct_slice_read](url)

It's sort of working, but the gcode generated includes a strange hashing pattern; I expect to see output on the left but I get output on the right

[image]

any suggestions?

BagelOrb, 2016-02-26:
Yes, update your json files to the latest master branch. That's all.

BagelOrb, 2016-02-26:
You should retrieve it from the Cura repository.

Ghostkeeper, 2016-02-26:
The infill-overlap setting was changed from being a percentage of the wall line width to simply being a distance in mm. So if you had an overlap of 10% that is now a distance of 10mm. You are slicing with a version of CuraEngine that has the update, but a JSON file which doesn't have the update yet.

codehero, 2016-02-26:
OK good to hear it's a simple fix, but this is where I got my file:

`https://github.com/Ultimaker/Cura/blob/master/resources/machines/fdmprinter.json`
It's only 14 days old. Is there a different place to get these json files or is this the wrong json file to use?

BagelOrb, 2016-02-27:
That's the right place. I've changed the infill overlap recently, so it
might well be possible that it's out of date..

codehero, 2016-02-29:
@BagelOrb I see, is there a location where I can download the updated files?

BagelOrb, 2016-02-29:
No the github link you posted is the right place; I'm just saying that 14 days might already be too old.

codehero, 2016-02-29:
The link points to a file that is 17 days old now. That link points to a file that is out of date; i.e. contains an infill with % units. Is there an update coming to that file soon?

BagelOrb, 2016-02-29:
Oh I'm so sorry. The json is indeed wrong. It did get changed to mm instead of %, but somehow it got changed back. 

I will fix this asap tomorrow.

BagelOrb, 2016-03-01:
I think I forgot to push the commit :/
Now the file does have an infill overlap in mm though.

codehero, 2016-03-01:
Thank you. After pulling that file and the recent changes made to CuraEngine my slicing is looking a lot better now.

BagelOrb, 2016-03-11:
How is your feature of directly sending slices to the engine going?
Did you get it to work?

Do you want to create a pull request, so that it will be included in the engine indefinitely?

codehero, 2016-03-11:
I am able to generate gcode from my slice data, but the slice parameters (layer_0_height, layer_height, etc) are ignored. I have not been able to test on actual 3d printer yet but I am still working on it in bursts.
My plan is to print in 2 colors, so I would want to test that before issuing a pull request.
Does geometry data for dual printing require a nonzero distance between the volumes, or can the volumes share vertices?

BagelOrb, 2016-03-12:
Did you implement it as an extra feature, or as replacing normal STL
slicing?

Of course those parameters have to be ignored.

I don't think there is any interaction between meshes in that sense. Also:
Note the setting multi_extrusion_overlap!

codehero, 2016-06-29:
I suppose down the road it could make sense to make STL slicing just a special case of a generic slice stack. For now I have the features side by side. I have been testing this code recently on a FlashForge dreamer and had a successful print in PETG.

I pulled to the latest master and have an issue on the perimeter layers of the print. The first image shows the good, the second the bad:

[image]

[image]

I am using the fdmprinter.ref.json as my settings. Any idea why the perimeter would turn into dashed lines?

BagelOrb, 2016-06-29:
If you've pulled today or yesterday you might be experiencing a bug with topbottom skin which I just solved.

In any case this looks like a visualization bug. Please check out how it looks in other software, like repetierHost.

Why are you merging to master? master is the development branch which has quite some bugs.

Also the current master is already miles away from the 2.1 release.
Now we don't have Cura/resources/machines/fdmprinter.json ,but Cura/resources/definitions/fdmprinter.def.json which is also quite a bit different in it's format.

codehero, 2016-06-29:
Sorry I had been out of the loop for awhile; I suppose I don't have a good reason to merge to master. In any case, it's definitely not a visualization bug as the part does not come out right. I did mention I was using fdmprinter.ref.json.

For this case, I will try a rebase to 2.1 to see if the performance is any better. I would like to maintain a working copy for the master though.

BagelOrb, 2016-06-29:
If you merged to master today then you should merge again, because I just
solved a bug with the skin area calculation.

However, that bug shouldn't result in this visualisation, so I suspect
there visualisation good haywire when it encounters my big in the gcode.

Any way merging to master now will be better than merging to master this
afternoon ;)

codehero, 2016-06-29:
I applied the patch to my master working copy and still see the same issue. As you can see, some of the lower layers come out fine but all the upper perimeters are fragmented

[image]

BagelOrb, 2016-06-29:
Ok. Thanks for checking.

I've never seen such a bug before. Perhaps it's related to coasting. Did
you enable coasting?

Will check tomorrow whether it also happens in master itself without your
edits.

codehero, 2016-06-29:
Coasting is not enabled. II just dove into the gcode and it is very strange; the extrusion dimension does not seem to be keeping pace at all with the x/y displacement. Machine nozzle width is 0.4; material diameter is 2.85.

;FLAVOR:RepRap
;TIME:6666

;Generated with Cura_SteamEngine master
M104 S210
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
;LAYER_COUNT:100
;LAYER:0
M107
G1 F1500 E-6.5
G0 F7200 X-17.4 Y14.7 Z0.3
;TYPE:SKIRT
G1 F1.5e+03 E0
G1 F1800 X-17.9 Y14.1 E0.0099069
G1 X-18.3 Y13.5 E0.019328
G1 X-18.8 Y12.9 E0.028751
G1 X-19.2 Y12.3 E0.038538
G1 X-19.6 Y11.6 E0.047817
G1 X-20 Y11 E0.057512
G1 X-20.3 Y10.3 E0.06693
G1 X-20.7 Y9.61 E0.076695
G1 X-21 Y8.94 E0.085902
G1 X-21.3 Y8.22 E0.095706
G1 X-21.5 Y7.51 E0.1052
G1 X-21.8 Y6.79 E0.11467
G1 X-22 Y6.05 E0.1243

Eventually we get:

;LAYER:5
G0 X-8.08 Y11.9 Z1.3
;TYPE:WALL-INNER
G1 F3.6e+03 X-8.47 Y11.6 E105.47
G1 X-8.85 Y11.4 E105.48
G1 X-9.24 Y11 E105.48
G1 X-9.59 Y10.7 E105.49
G1 X-9.95 Y10.4 E105.49
G1 X-10.3 Y10.1 E105.5
G1 X-10.6 Y9.72 E105.51
G1 X-10.9 Y9.36 E105.51
G1 X-11.2 Y8.99 E105.52
G1 X-11.5 Y8.62 E105.52

Which explains the staggering: with no apparent change in the E dimension, there is no line to render.

BagelOrb, 2016-07-04:
It seems like the E-values always have 5 non-zero digits, rather than 5 after the decimal dot. It seems you are writing output in sdt::scientific format, rather than std::fixed.

It seems you are not specifying an output file while slicing via the command line. I never really tested it, but I've solved the problem for you in f3b6042243bc41591a96dbf3015b25cf09a7fbf5

codehero, 2016-07-07:
Tim, that eliminated the precision errors. However, I still need to fix how much material is extruded as it is way too little. I think I just need to play with the settings a bit more.

Also I am printing to stdout and redirecting, not specifying a file on the command line. Is this something to be avoided?

BagelOrb, 2016-07-07:
That should be possible, but it hasn't been tested for over a year, so
please report bugs with that if there are more.

BagelOrb, 2016-07-07:
Perhaps your line widths are (visualised) wrong because you are using a (non)volumetric gcode flavour for visualisation/gcode generation.

It might also be due to the filament diameter.

Ghostkeeper, 2016-07-08:
I've used the redirecting-to-file technique quite often recently. Works fine.

nathanielvirgo, 2019-11-10:
I couldn't tell from this thread: did this feature get merged in, and is it currently available in Cura? I would like to print some mathematical surfaces, and it's much much much easier for me to generate slice data from them than a mesh. If this feature is available it would be super useful. Are there instructions for using it somewhere?

BagelOrb, 2019-11-10:
Nope. It's still open. Mathematicions don't have priority I guess. ':D

BagelOrb, 2019-11-10:
Note that generating the mesh doesn't need to be as difficult as you might think.
You don't need to connect the layers together, you only need to give your slices a height, so that Cura will slice each layer exactly where you want.

BagelOrb, 2019-11-10:
Just generate the 2D polygons and then extrude each line segment into two perfectly vertical triangles with the same height as your intended layer height - and BAM! you got what you want.

BagelOrb, 2019-11-10:
(cleaned up discussion of unrelated stuff)

codehero, 2019-11-11:
For the OCD people, can you quickly explain how making a non-water tight STL, with the possibility of internal cavities etc, would work with properly with Cura? Wouldn't support generation not work without horizontal faces?

nathanielvirgo, 2019-11-11:
BagelOrb, a question (related to codehero's comment): in order to do what you suggest, would I also need to triangulate the top and bottom surfaces of each slice and union them to make a manifold mesh? Or does Cura only look at the triangles that actually intersect with the slices (so it wouldn't even notice if the top and bottom were missing)?

(I know that the triangulation and union operations can be done in OpenSCAD, and having run some tests it can actually do it quite quickly, but I'd prefer to avoid the dependency.)

BagelOrb, 2019-11-11:
The first thing CuraEngine does is slice the 3D model into polygons and the rest of the processing only uses these polygons. The 3D model is *only* used to calculate the areas of each layer and all other operations are operations on polygons.

Your mesh doesn't need to be manifold; only each slice needs to be a closed polygon.

The slicing stage within CuraEngine is a 2 step process:
- for each triangle generate the line segments for each layer with which it intersects
- for each layer stich all line segments together into polygons

nathanielvirgo, 2019-11-11:
Great! I was just testing it as you were posting that message - I made an STL file containing only the sides of a cube, with no top or bottom, and it sliced it just fine, creating a top and bottom layer and infill etc. So this will be easier than I thought. Thank you for the tip!

BagelOrb, 2019-11-11:
Your model can look like this:
[image]

nathanielvirgo, 2019-11-11:
Perfect, that's easy to do.

Second question: how can I know the exact heights at which Cura will slice the model? For example, if I have my bottom layer set to 0.25mm and all other layers set to 0.2mm, would I need to make sure my model has layers at 0mm, 0.25mm, 0.45mm, 0.65mm etc., or in the midpoints of the layers, at 0.125mm, 0.35mm, 0.55 etc.?

BagelOrb, 2019-11-11:
They are sliced in the middle, unless you have the setting Slicing Tolerance set to Exclusive or Inclusive (in which case the layers are sliced at the top and at the bottom and a union or intersection is performed on those slices).

I'm not sure whether this is still the case but it could be that all layers are sliced 0.5 * layer height below their top, meaning that the initial layer is sliced slightly above the middle - assuming the initial layer height is higher than the normal one.

Ghostkeeper, 2019-12-03:
> I'm not sure whether this is still the case but it could be that all layers are sliced 0.5 * layer height below their top, meaning that the initial layer is sliced slightly above the middle - assuming the initial layer height is higher than the normal one.

No, the initial layer's slicing Z coordinate is set to `initial_layer_height / 2` when using tolerance Middle.

codehero, 2021-10-15:
@LipuFei I was able to achieve what I wanted by building horizontal triangle strips as they have shown.