Ultimaker/CuraEngine #311: Infill Wipe Distance not Functioning?

CCS86, 2016-02-26:
Here, I have it set to .35mm, so it should be very evident, but looks like no infill wipe at all:

[image]

Any ideas?

Thanks

CCS86, 2016-02-26:
Ah, looks like I had bottom layers cranked way up.

I'm assuming that infill wipe is disabled on bottom/top layers. Is this intentional?

BagelOrb, 2016-02-27:
Well it's kind of intentional - if that's a thing.

For infill the adhesion to the walls is more of a problem.

Do you think it would be better to have it on the skin infill as well?

Or would it be better to introduce yet another setting for skin wipe?

CCS86, 2016-02-27:
I think it's helpful during solid infill as well. It helps to fill in the
tiny "stair step" gaps between adjacent infill passes, without commanding
extra extrusion via infill overlap.

I think it could just be applied via the same parameter. Or if some are
very opposed to it, maybe a check box to disable wipe on skin.

BagelOrb, 2016-03-01:
Fixed via the same parameter.

closed 2016-03-01 by BagelOrb