Ultimaker/CuraEngine #312: Infill went crazy in the middle of the object

ynilsiam, 2016-02-26:
Hi,

I set the pattern to be concentric.
Everything was good until in the middle the pattern change to be crazy, like xxxx back and forth.
Any suggestions?

Thanks,

nickthetait, 2016-02-26:
Can you post a picture of it going crazy and the STL that you used?

BagelOrb, 2016-02-27:
What version of CuraEngine are you using?

Why are you using concentric infill?

ynilsiam, 2016-02-27:
Hi,

I am using 15.0.6.
I am not sure what happen, now it does not go crazy any more. I tried to
change the line_width and layer_thickness and other things.
However, in the middle it does not stick with the concentric pattern. It
tries to fill the gap I guess.
Is there any way to make it stick with the concentric pattern, not zigzag
pattern?

Thanks,

BagelOrb, 2016-02-27:
I advise to stick with cura 15.04 until cura 2 is fully released. The 15.06
version was a beta which was quite buggy.

closed 2016-03-07 by BagelOrb