Ultimaker/CuraEngine #314: Horizontal and Vertical Offsets

MicheliDE, 2016-03-01:
Hi,

I posted this first on the ultimaker forum, they referred me to here to ask the question and describe the situation

I have been recently using my ultimaker for precision part sizing and have been struggling with part fits. Due to shrinkage of material based on a number of factors I have been looking for a quick and easy way to adjust for fit. I have found the Horizontal offset works for holes and I believe it may work for all inner surfaces also?

My issue is that I have a complex part that I have had to slice in two to print accurately some of the features, I had to rotate it because I cannot print it flat with the top down due to some features and overhangs and fanning requirements. I had to print each of the two halves separately and glue them together later.

I am trying to find a way to vertically offset holes in the Z direction as well as inner surfaces in the same direction. At first I thought that I may be able to use the Horizontal offset and it would offset the base inner surfaces all the way up to the point of the arch where it joins together but I am not sure if it actually able to perform this operation and it doesn't help with the fit of the overall thickness of the half of the ring.

Does anyone have suggestions?

Ideally I would like to see offsets that affect either the inner or outer walls of the model in XY and Z separately. But I don't believe this is yet part of Cura.

MicheliDE, 2016-03-01:
[image]

BagelOrb, 2016-03-02:
Mathematically speaking all faces are on the outside in your model. Only
hollow objects have faces which you can call inside.

Performing the XY offset is performed on the 2d polygons after the model
has been sliced. Performing an offset in 3d is rather difficult. If you
want that you should try to do so in your modelling program.

Why don't you just print your part with the flat piece on the bed? That way
there is no overhang anywhere...

Ghostkeeper, 2016-03-02:
OpenSCAD has a [Minkowski sum](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#minkowski) feature which can perform the XY offset too. You can do the Minkowski first and then rotate it afterwards. It becomes a bit of a hassle though if you have to do this repeatedly for tweaking. There is an API into OpenSCAD where you can run arbitrary OpenSCAD scripts and get an STL back, but I've never used it so I don't know how it works.

MicheliDE, 2016-03-02:
Hi Tim,

Thanks for the reply. I am trying to build a very precise part.

I have thought through different ways to orient the whole part on the build
plate and there isn't a way that will achieve a smooth underside, as you
know overhangs dip slightly no matter what and using suspension or towers
does not create a smooth underside when you break them away. This is why I
originally divided the part into two pieces so that it could be glued
together later using the flat of the build plate (this creates an almost
invisible seam). Even if I did use suspension or towers there would be some
serious trouble printing the holes properly

Also the center where the holes are has a slightly convex shape and there
is a feature that goes around the outside of the top. The sides of the part
have draft angles as this will eventually be an injection molded part. I
need to test the acoustic properties and this is a part of an assembly so
everything needs to fit together really well. Unless you can think of a
better way to print the part I believe this way creates the smoothest part
on all surfaces. The only issue I have is the wall thickness which is not
coming out close enough due to part shrinkage.

So as you say above I would have to have an enclosed shape for Offset to
work on it so Offset would work on the holes or it would work on the entire
part if it was one piece and if it was horizontal rather than vertical. If
the same rule applied to Vertical offset I would also need the ability to
use the build plate as the closing side of the arch for printing half of
the overall part.

Here are some more images to give you some additional insight into what I
described above.

Best Regards,
Dean

MicheliDE, 2016-03-02:
I guess the images didn't appear here from email so here they are--
[image]
[image]
[image]
[image]

MicheliDE, 2016-03-02:
Hi Ghostkeeper,

Thanks for the response.

With the Minkowski function does it take input as a dimensional offset such as mm like cura does or does it use some kind of proportion? Reason why I ask is that I have set up a spreadsheet that compares Cad Dimensions to printed part dimensions to record the offset required.

Does Minkowski apply the offset only to closed features or can it apply it to an open feature or part thickness? This model cut in half has an open feature as it is cut into two parts for printing. I suppose I could slice it after performing the function but this adds another layer of complexity. Also does Minkowski apply this to only inner hole like features or can it apply it to outer part features as well?

Best Regards,
Dean

MicheliDE, 2016-03-02:
Here is some image to try to further define the issue:

In Application
[image]

Defining Terms
[image]

Ghostkeeper, 2016-03-03:
If you take the Minkowski sum of your model and a circle, the result would be the union of that circle placed at every point on that model. Take for instance the sum of these two triangles:

[image]

This means that the Minkowski sum would dilate the holes as well as the outside of the model. How much it dilates is exactly the radius of the circle, so an absolute distance like millimetres, not proportional. It'll work almost the same as the X and Y offset in Cura, except that summing your model with a circle would produce rounded corners rather than a miter. A sum will make the holes smaller. To make them larger you'd need to do a Minkowski subtract.

If the model is not watertight due to your splicing, I think OpenSCAD would give an error that the model is non-manifold. You could also do the splicing itself with OpenSCAD. It has functions to take the volumetric difference between two models, so you can simply subtract a cube from your model that is positioned such that the cube overlaps with exactly half of your mesh, and it will return a mesh that is watertight and correct.

The Minkowski sum would also dilate the side where it was sliced. This can be resolved by doing the Minkowski sum before splicing in two, or subtracting a cube in the same way as the splice I described above, by subtracting a cube on the side where it shouldn't have dilated.

OpenSCAD is very much a programmer's dream. It can automate a lot of stuff, like first dilating and then splicing, rotating it correctly for printing, all that sort of things. But to do that it uses a fairly powerful programming language. It's not hard if you're used to programming but for others it will take some time to get the hang of it.

To sum up this rather lengthy post, I see two options for using OpenSCAD in your process:
* First splice the model, then have OpenSCAD perform the Minkowski sum in order to dilate the holes, then have OpenSCAD subtract the unintended dilation on the side where you spliced. Then send the result to Cura to slice.
* First have OpenSCAD perform the Minkowski sum in order to dilate the holes. Then have OpenSCAD splice the model. Then send the result to Cura to slice (better option in my opinion).

MicheliDE, 2016-03-03:
Hi Ghostkeeper,

Thanks for the information. I can see how this could work and I assume that it will perform the same operation on the holes and inner surface of the outer ring equally because it applied the function to the whole of the object?

Here is my concern. The outer walls are not 90 degrees to the top of the part. Yes they are for the holes but not for the walls. The walls have a draft angle. If I perform the Minkowski subtract is it going to subtract my draft angle?

Best Regards,
Dean

Ghostkeeper, 2016-03-04:
Hmm, well it wouldn't affect the draft angle, but it would make the wall thicker in total. And that's something you don't want either then I suppose.

MicheliDE, 2016-03-04:
Correct, I would like to be able to either extend or shrink the insides or outsides of the whole object for compensation of material shrinkage or expansion during printing, and to be able to do this for all three axis and select ably able to do this with an open side that is connected by the build plate.

I am sure that this is very tough to add on in code or with a single formula. Originally I thought that I could use the Horizontal expansion in conjunction with XYZ scaling of the object to fit the part, and for two other parts I was able to perform the function to the precision that I needed. It is just the Z axis due to the splitting of the part and need to print that is causing parts to not fit with no solution other than to redesign the part for printing, then again for manufacturing.

MicheliDE, 2016-03-04:
[image]

MicheliDE, 2016-03-04:
Horizontal offset is working and does not affect draft angle, as the part is build higher off the build plate the 2D offset becomes less effective. The image shows a part with the offset applied VS another part without the offset.

[image]

closed 2016-10-10 by BagelOrb