Ultimaker/CuraEngine #316: MACH3 Gcode flavor is using wrong parameter prefix in M codes

kniftar, 2016-03-05:
MACH3/LunuxCNC requires "P" as a prefix for mcode parameters.  CuraEngine is currently using "S" when MACH# is selected.  Should be:
M109 P#
M104 P#
M190 P#
M140 P#

BagelOrb, 2016-03-05:
Ok, good to know.

I don't think we will fix this any time soon, cause supporting third party
printers isn't exactly high on our priority list.

However, if you would fix the code, we can incorporate the fix into Cura.
We're always up for considering a pull request.

Ghostkeeper, 2016-03-07:
It could also be implemented as a post-processing script in Cura.

The alternative is going through CuraEngine and making the M-codes it writes dependent on the `machine_gcode_flavor` setting.

kniftar, 2016-03-09:
Right,  if I get some time I would like to try and help fix this but never using github or contributing to a project it seems like a bit of a learning curve.  I have a little C programming experience but no cpp.  That being said it looks like a simple fix. I do see where the Mcodes are written in gcodeExport.cpp and for what it's worth the G04 (Dwell) is also different for mach/machinekit/linuxcnc.  It's parameter is seconds not milliseconds.

BagelOrb, 2016-03-09:
You should create a fork from this github. In the *fork* you van edit
whatever you want. Then when you're done, create a *pull request* to submit
your code to us.

Google the last step if you don't know how.

There is an option to not depend on arcus, which should make the build
process a lot easier ;)

Good luck!

kniftar, 2016-03-12:
errrr, I'm not having much luck building from source and I usually can figure this stuff out.  I've tried cura-build to make a package and also the ubuntu15.04 install script in the cura-build repo. I've tried both on new vanilla installs of ubuntu 15.04, 15,10 and 16.04.  Seems arcus is the spoiler and I don't see the option to remove the dependency.  Could anyone share a recipe for this that works?  Thanks much..

BagelOrb, 2016-03-22:
Set ENABLE_ARCUS to FALSE and CMAKE_BUILD_TYPE to RELEASE

BagelOrb, 2016-03-22:
Also you can set BUILD_TESTS to FALSE

Renha, 2016-04-13:
> There is an option to not depend on arcus, which should make the build
process a lot easier ;)

woooooow! and all would still work?

BagelOrb, 2016-04-14:
That should work. I don't know if the option is still intact after the
merge from 2.1 to master, but hopteelt it is!

Note that without arcus the engine cannot communicate with the frontend...

BagelOrb, 2016-10-11:
The whole platform is below Z=0...

Renha, 2016-10-11:
oh, I see. Well, building without arcus works, i've just checked. Package for Arch is cura-engine-noarcus-git at AUR in case if someone need it.

BagelOrb, 2017-07-23:
outdated

closed 2017-07-23 by BagelOrb