Ultimaker/CuraEngine #317: Feature Request: Ability to play a sound when a print is complete

starr0stealer, 2016-03-10:
The end user can have the option to define a sound to be played once a print is completed. There can be built in sounds to select from, or allow the user to pick a sound file on the system. For printers with a heated bed the sound could be delayed until a target temperature is met (or multiple sounds can be played during each stage).

This would assist with longer prints that the user doesn't stay by the printer during the entire job, providing a notification that could be heard from a distance.

Could be handled via the plugin system (once its expanded to handle UI or other features outside of post-processing) or within the core software (preferred option as this would happen sooner than later).

BagelOrb, 2016-03-11:
You reported the issue in the wrong github page. You should report it to
Ultimaker/Cura

Fun idea though.

starr0stealer, 2016-03-11:
My apologies I didn't notice the other project, came here from the plugin documentation

closed 2016-03-11 by starr0stealer