Ultimaker/CuraEngine #318: quesiton - installing legacy

pkTux, 2016-03-10:
From octoprint documentation, legacy is still required. i Git downloaded it & tried to follow the regular cura make instructions... those fail to find 'cmake' files or something.
Does anyone know the process for compiling legacy? 
If its needed, i am using ubuntu 16.10

Ghostkeeper, 2016-03-11:
As far as I know, you merely have to switch to the [legacy](https://github.com/Ultimaker/CuraEngine/tree/legacy) branch of CuraEngine, go into the terminal and hit `make`.

Ubuntu 15.10 is known to give problems with compiling modern versions of CuraEngine sometimes due to a conflict with Protobuf, but with Legacy that shouldn't be a problem.

pkTux, 2016-03-11:
ghost, for what its worth i tried that & got a make error. but then i checked & realized that ubuntu has a cura engine in its repos thats old enough. Thanks for the help.

closed 2016-03-11 by pkTux