Ultimaker/CuraEngine #321: scenegraph for scene management

sajjadul-wematter, 2016-03-21:
Hello folks,

Is there any online for this API to disscuss some relevant issues? I did not see any , therefore I am posting it here. How do you manage a large scene that has to fit inside the printing region ? In other words,  If there are several copies of the same model, do you save and send all the coordinates of each model to slicer,, or you manage the scene using the scenegraph data-structure first , then slice it and eventually generate gcode for each of the copies of the model ?

Thanks

nallath, 2016-03-21:
The Cura GUI (conveniently found at https://github.com/Ultimaker/Cura) uses a SceneGraph to do this. 
If there are multiple copies of the same model, it's sent multiple times. G-code is generated for the entire scene in one go. We would like to move towards g-code per model (as it would save a lot of time when moving a single object).

sajjadul-wematter, 2016-03-21:
Thanks for the response! Let me understand first of what you mentioned through an example. I have created several copies of the very same model well-aligned inside the bounding box of the GUI. Before the printing commands are to be sent, each of the copies has to be sliced and generate the tool-path for each of the copies of the model. What I am assuming to do is that, you declare and arrange the scene in a scenegraph data-sturcture, where only one reference is made to the model geometry because of their equivalence and that geometry is referenced by the rest of the other copies by translation nodes within the data-structure. Then we slice only the main geometry and rest of the copies are just affine transformation of the main geometry. Eventually G-code is generated for all of the copies in one go. I did not understand the part you mentioned  - "If there are multiple copies of the same model, it's sent multiple times." Where is it sent multiple times ? is it sent to the slicer ? Some explanation will be helpful to understand the concept.

nallath, 2016-03-22:
Yes, it's sent to the slicer multiple times.

Ghostkeeper, 2016-03-22:
The way that data is sent to CuraEngine is quite naive at the moment. It makes actual copies of the vertex data, transforms those on the Front-End side, and sends them off multiple times to CuraEngine.

sajjadul-wematter, 2016-03-22:
Do you think that the idea that I presented will be optimal for future Cura Engine ? Some guide line would be very helpful.

BagelOrb, 2016-03-22:
It is a bit unclear to me what idea it actually is that you are presenting, but it certainly would be nice to send over the models only once.

If CuraEngine wouldn't close after each slicing, we could send the model data only once and resend the scene-graph for each slicing. Then we could apply your trick of copying the polygon data generated by FffPolygonProcessor and transposing it.

The only thing is that I think the use-case is quite small, since it only applies of the clone is translate only in the xy-plane and isn't rotated, scaled or translated in the z direction.

Other than that the idea of keeping the models loaded in the engine has a number of other benefits.

sajjadul-wematter, 2016-03-23:
The idea is that , if we have multiple copies of the very same model, we can store only one main geometrical reference to the model and rest of the copy will refer to the main reference model using scenegraph datastructure. Each of the copies of the main reference model contain only the information related to its affine transformation. Then We send only the main reference model to slicer to slice it and rest of the copies are affine transformed reference of the main sliced model. Then we generate the toolpath for the main sliced model and the toolpath is also transformed for each of the copies. Eventually Gcode is generated for each of the copies of the model. I hope that I made it clear this time. Instead of sending the model multiple times to the slicer , I am suggesting to send it only once. I shall be looking forward to your feedback over this issue.

BagelOrb, 2016-03-23:
The part of your idea about sending the models and scene graph separately does interest me, but the part about copying the resulting toolpath and performing affine transformations on those copies won't work.

When a copy of a model is translated or scaled in the z direction, the layers should come from different parts of the 3D model than the original. Simply applying affine transformations will never get you the gcode you should have. The only affine transformations which don't have this problem are translations purely in the X-Y plane.

Given this limitation I think the trouble to implement it outweighs the benefits.