Ultimaker/CuraEngine #324: Slicer Question

KJHamblin, 2016-04-04:
I'd like to learn more about how slicers work and I wanted to ask where a good place to begin is.  I've downloaded the source code and looked through it but comments are scarce and if I found the correct wiki, it was short and more of a general description.

KJHamblin, 2016-04-04:
I'm particularly interested in how the layers get translated to gcode and when the position of the model on the build plate matters.  Does the slicer know the x,y when it is creating layers or did it only use the x y when making gcode.

Thanks for any help and direction!

BagelOrb, 2016-04-04:
The model contains the x y and z data. The model then gets sliced into
layer outlines. From these outlines we create walls by performing offsets
on the outline polygons and the skin and infill areas by performing boolean
operations on the polygons.
In fffgcodewriter this layer information is converted to paths by finding
the optimal order in gcodePlanner, which converts the paths to gcode.

Hope this helps.

KJHamblin, 2016-04-05:
Thanks for the reply, that does help.

The x y data I was referring to was the model's position on the build plate.  I wasn't sure if the slicer uses that information when it slices or if it only uses the position on the build plate when it is creating the paths.

BagelOrb, 2016-04-05:
That data is already present in the data the engine receives. The frontend
sends the transformed mesh data, not the original mesh data and the
transformation...

It's possible to do it that way though, if you want to slice directly via
the command line for example.

sajjadul-wematter, 2016-05-23:
I am also going through the slicer code and I have the following question:

1. 	int32_t layer_max = (maxZ - initial) / thickness;

My interpretation:

maxZ - the height value of the triangular primitive that is calculated in the previous steps.
intial - NOT SURE ABOUT ITS PURPOSE - all that I could extract is that it comes through the constructor. Some explanation will be helpful.

thickness - the height of the layer along Z-direction.

Basically the statement is calculating the maximum number of layers to loop through or number of layers to create. 

2.         for(int32_t layer_nr = (minZ - initial) / thickness; layer_nr <= layer_max; layer_nr++)
            {
                /*
                    the initialization of the for loop is confusing for me. 
                */
            }

Thanks

BagelOrb, 2016-05-23:
>    1. for(int32_t layer_nr = (minZ - initial) / thickness; layer_nr <=

The amount of layers depends on the height of these initial layer and the
distance between consecutive layers.

When slicing an object which has a height of 2cm with an initial layer
height of 1cm and layer height of 1mm, there will be 11 layers.

Where in the code are you referring to exactly?

closed 2016-10-10 by BagelOrb