Ultimaker/CuraEngine #325: Perimeter Seam Overlap

CCS86, 2016-04-07:
Hi guys,

I'm curious how the perimeter loop end point is calculated relative to the start point. Does it stop one nozzle diameter short of the start? Or are you doing a calculation of overlap where the area of the overlapping circles is equal to the area of the two dimples above and below?

[image]

Either way, I think loop overlap would be a great parameter to add. Currently, it is not bad at all. The seam is quite small. But it is a slight dimple, and I think by tuning the overlap, you could almost get it to vanish.

BagelOrb, 2016-04-07:
The overlap is currently simply 100%.

The lines are modeled as rectangles, rather than rounded rectangle which
means that there will be half a circle of underextrusion at the start and
half a circle of overextrusion at the end of a perimeter..

CCS86, 2016-04-07:
So, you mean that the start and end points are exactly the same?

BagelOrb, 2016-04-07:
Yup.

CCS86, 2016-04-07:
That doesn't seem to add up. If the start and end points are the same,
there would be a full circle of overextrusion, and likely a bulging seam in
the print.

BagelOrb, 2016-04-07:
It's not that bad as prints are right now.

I compensating feature would look something like coasting; it would
probably be computed within the same algorithm...

CCS86, 2016-04-07:
I'm not sure I'm understanding. 

To clarify my experience, the print quality on this build is outstanding! I'll take some macro pictures soon of my last print that shows this fact off. Gap filling is working great. I'm using different line widths and sped for perimeters and infill, and the extrusion is extremely reliable and consistent. 

Even the perimeter seam is very small... but slightly puckered inwards. Coasting does not seem to have a noticeable effect on this for me. So, I really think a small adder to the end point of the loop would allow fine tuning of the seam.

daid, 2016-04-08:
IMHO, CCS86, I think you are wrong. As your base assumption is wrong.

Extrusions put down are not round like that. If they where, you would have a 100% overextrusion at the start/end point. But we clearly see that that is not the case.

CCS86, 2016-04-08:
My base assertion is that adding a parameter to control perimeter loop overlap, would allow fine tuning the printed seam, and be very beneficial. Everything else in this thread, is my trying to understand how the current algorithm works.

Really, what the cross-sectional shape of the extrudate is, viewed on the XY plane, is inconsequential. What is important, is what do finished prints look like.

I am printing very well tuned prints with this latest build, and still have a small dimpling at the seam. Yes, in an ideal world, starting and stopping the perimeter loop in the same position, would cause overextrusion. But in reality, there is a lag in extrusion at the start, and a lag in print head motion at the end, that end up leaving a small amount of underextrusion.

The calculation could evaluate the slope of the first perimeter segment, using that direction, and the length from the "overlap" parameter, to generate an additional move, before retreating inwards (it could just be a coast move I'm sure).

CCS86, 2016-04-08:
Here is an example of what I'm talking about:

73* overhangs without support:

[image]

Seam pucker:

[image]

CCS86, 2016-04-27:
Does anyone else have an opinion?

Ghostkeeper, 2016-04-28:
I've rarely seen it as severe as in that photo, but "seam pucker" is happening on my prints. Goes back quite a long time. I don't rightly know how easy or safe it would be to fix it though. Just my €0.02.

BagelOrb, 2016-04-29:
Let's bypass the whole discussion here; it's getting a bit out of hand.

I am not convinced your idea will work, but I'm also not convinced it will not work.
We should have a proof of concept, before I'm going to implement it.
If you edit your gcode by hand such that the feature would be present for the first half of that one file and the eventual print looks better on the lower half then I'm willing to consider implementing the idea.

CCS86, 2016-04-29:
Good call. I can work on that.

I program 5 axis mill-turn machines, but haven't dug too much into reprap gcode. 

Are any logical operators supported? ie: IF, THEN, GOTO

Any math? ie G1 X=4+.2

Thanks!

BagelOrb, 2016-04-29:
Nope. 3D printing gcode is pretty straightforward. Have a look at the gcode
generated by cura.

You shouldn't generate gcode, though. I suggest you modify existing gcode...

daid, 2016-04-29:
RepRap GCode has nothing in common with advanced CNC milling gcode. Every feature you are used to, forget it.
3D printing gcode is a simple instruction list, nothing more. No loops, no conditions, no variables, no math.

DDDirk, 2016-04-30:
Of course you could write a program to write the gcode. I did that to print tensile bars, where I encountered the opposite problem, getting a blob in the corner where the printing of the perimeter started and finished. Then I realised that the paths should not have the same start- and endpoint, because the laid down road has dimensions. I solved this by putting the endpoint half a nozzle width back. Not a whole nozzle width. So I created the overlap suggested in the first post. Worked just fine. And it should help solve the dimple/seam/pucker problem, although tweaking the overlap for different shapes might be needed.

BagelOrb, 2016-04-30:
I'm just saying that it might be easier to perform some smart regex search
and replace.
Generating the gcode you may encounter problem which have been solved in
CuraEngine already. No reason to reinvent the wheel ;)

Of course you could write a program to write the gcode. I did that to print
tensile bars, where I encountered the opposite problem, getting a blob in
the corner where the printing of the perimeter started and finished. Then I
realised that the paths should not have the same start- and endpoint,
because the laid down road has dimensions. I solved this by putting the
endpoint half a nozzle width back. Not a whole nozzle width. So I created
the overlap suggested in the first post. Worked just fine. And it should
help solve the dimple/seam/pucker problem, although tweaking the overlap
for different shapes might be needed.

DDDirk, 2016-04-30:
Well, 'the wheel' (Cura) apparently solved this problem somehow. If indeed the paths have the same start and end point, then something else apparently prevents the blob from appearing. And that same thing might then be tweaked to prevent the seam from appearing.

CCS86, 2016-04-30:
Yup, I was planning on using Cura code and some regex replacement to automate a bit.

@DDDirk, I think that the seam bulge might have been solved with an immediate rapid "retreat" move at the seam endpoint, to get the head away from the surface before doing any retract or hop moves that cause a dwell.

DDDirk, 2016-04-30:
What do you mean by 'retreat'? Move the nozzle back a bit?

But when I slice a cube with Cura, I see nothing special happening.

G0 F9000 X120.899 Y102.101
;TYPE:WALL-INNER
G1 F3000 X120.899 Y120.899 E356.82913
G1 X102.100 Y120.899 E357.58109
G1 X102.100 Y102.101 E358.33301
G1 X120.899 Y102.101 E359.08497
G0 F9000 X121.299 Y101.701
;TYPE:WALL-OUTER
G1 F3000 X121.299 Y121.299 E359.86889
G1 X101.700 Y121.299 E360.65285
G1 X101.700 Y101.701 E361.43677
G1 X121.299 Y101.701 E362.22073
G0 F9000 X120.299 Y102.701
;LAYER:7
G0 X120.758 Y102.687 Z1.000
;TYPE:FILL

The paths for both walls end where they started, followed by a G0 to the next wall or the infill.

About those tensile bars of mine, forget about that, I was experimenting with all sorts of alternative ways of slicing (which is why I programmed it), so there may have been all sorts of causes. This just caught my eye because what you propose is very much like what I did there.

BagelOrb, 2016-05-01:
If you print two cubes it will do a retreat, ie move inside the printed
part..

CCS86, 2016-05-29:
Okay, finally got a chance to test this out, and the results look promising.

I printed 4 tests:

A control, with coasting disabled, as output:
[Test-NoCoast.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/288592/Test-NoCoast.gcode.txt)

Coasting enabled @ 0.2mm^3, min_volume: 2mm^3, as output:
[Test.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/288594/Test.gcode.txt)

Modified gcode, where I inserted an additional coasting segment, at the end of each loop, that continued 0.2mm beyond the normal start/end point, in the direction of the initial move:
[Test-mod.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/288593/Test-mod.gcode.txt)

Same as above, but 0.4mm (nozzle diameter):
[Test-mod2.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/288591/Test-mod2.gcode.txt)

Results:

[image]

CCS86, 2016-05-29:
I think that I could push that advance just a touch further, and it would be about as good as you could get. Then, you flip it to shortest/random, and I bet they essentially disappear.

CCS86, 2016-05-29:
Also interesting is the inside loop. Since this was a solid object that I printed with 0% infill, I would guess you normally never see this surface, but it illustrates two useful things. Beds oscillate during layer change. And without extruder advance, the start of print segments, especially with a speed increase, give you a period of under-extrusion.

 I have the UM heated bed kit, and felt it wasn't stiff enough in bending. I bolted twin 0.5"x0.5" angle stock along the length as frame members. This stiffened it a lot, but it still can affect print quality. These are only 0.1mm layers, so the effect is worse with thicker layers.

These prints were starting around Y60, so I put a 90mm shift into Y and did it again. Just moving to the back of the bed, as I usually do, reduces the effect a lot!

[image]

Ghostkeeper, 2016-05-30:
Huh, so if I understood it well, coasting gives under-extrusion on the inside bend there?

Great to see that it has some positive effect on the outside at least.

CCS86, 2016-05-30:
I think it has more to do with the speed change and bed bounce.

The start of every single lap is after a coasting move. So I can't see why
it would affect the inside, but not the outside.

CCS86, 2016-05-31:
What say you @BagelOrb ?

CCS86, 2016-06-10:
I know you guys are super busy, but I think that reducing perimeter seam visibility is a pretty fundamental and beneficial pursuit.

It doesn't seem like the implementation would be super difficult (I could be wrong), and my results suggest enough benefit for at least an "experimental" setting for some additional testing.

BagelOrb, 2016-06-10:
What I'm thinking right now is a feature to add a small travel move after printing the outer perimeter to close the loop better. Is that a good idea?

CCS86, 2016-06-10:
I really think so, and my n=1 test agrees.

I'm sure you have a better idea on how the actual implementation would go, but my thought is to look at the slope of the first print move (or first x print moves), of a perimeter loop, and use that to generate an additional travel move and the end of the loop, with a user configurable distance value. My test looks pretty great when that distance is equal to nozzle width, but once it is implemented in the engine, it will be much easier to test different values, and see if other relationships are important (like speed or layer height)

BagelOrb, 2016-06-10:
I've logged it in our internal issue tracker.

CCS86, 2016-06-30:
That is odd. It sounds like what you saw is at odds with my observations.

You can see from my pictures that I don't have any appreciable over-extrusion, and that adding a small amount of seam overlap almost completely made the seam vanish.

CCS86, 2016-06-30:
Have you tried printing my test gcode files? 

I think a 1:1 comparison like that would be very valuable. I know that my UM is pretty heavily modified, so it would be great to looks for similarities / differences between different machines, to see if seam overlap has a similar effect.

BagelOrb, 2016-07-04:
I will try to find some resources to do these prints.

I agree that a comparison is quite valuable.

CCS86, 2016-07-04:
It will cool to see how universal or diverse the results are.

If anyone needs to print with 1.75mm filament, these files have a flow override command: https://dl.dropboxusercontent.com/u/60958586/Seam_Overlap_1.75mm.rar

CCS86, 2016-07-07:
Here are results from another type of machine that someone else got:

[image]

CCS86, 2016-07-11:
I added a 0.6mm overlap version to the archives. It has given me an even better result.

CCS86, 2016-07-15:
I just added a 0.8mm overlap file to the archive above. Haven't had a chance to test it yet.

@BagelOrb, Have you had a chance to test any of them?

CCS86, 2016-07-19:
Gcode files for 2.85mm filament (UM2):
https://dl.dropboxusercontent.com/u/60958586/Seam_Overlap_UM2.rar

CCS86, 2016-07-20:
I reprinted all tests last night to include the .6mm and .8mm overlap tests in my results. 

For me, I would say a 0.6mm overlap gives the best result.

[image]

CCS86, 2016-07-20:
On another person's UM2:

https://d2py9w124w2itd.cloudfront.net/photo/image/700x0/578f81d55145b/Seams.jpg

Ghostkeeper, 2016-07-21:
Hey, what's your line width in those prints? 0.35mm?

CCS86, 2016-07-21:
It's set to 0.4mm, which should be fine even with a .35mm nozzle.

CCS86, 2016-07-22:
I updated the UMO files with a 100mm Y shift, to push the print towards the back of the bed. Having less bed bounce helps to see the seam more clearly. I can do so with the UM2 files as well, but am not as sure about the bed dimensions.

CCS86, 2016-07-25:
Anyone?

Ghostkeeper, 2016-07-25:
We have plenty of test results from you, CCS86. Thank you for that. But the issue doesn't seem to be on our radar right now. In fact, I can't find it at all. @BagelOrb, you said you made an issue on Jira on the 10th of June. What number does it have? There's new photos that should be added to it.

BagelOrb, 2016-07-31:
It's issue cura-1698: outer wall wipe distance...

Ghostkeeper, 2016-07-31:
Ah, right, so scheduled for the release after 2.2 then.

BagelOrb, 2016-10-21:
Release 2.2 got renamed to 2.3 because of reasons (?)

I've implemented this for Cura 2.5.

It uses the normal travel speed for wiping off the nozzle after the perimeter. Would that be ok? I didn't know what to use else and I didn't feel like adding yet another setting for it...

CCS86, 2016-10-21:
Sound great!

CCS86, 2016-10-25:
Perfect, thanks!

Ghostkeeper, 2016-11-07:
This has now been merged. It should be available for 2.4.

However, my results weren't as nice as CCS86's. The g-code for it seems to be correct, but it makes no real difference in the seam. Perhaps it was because I was using silver-grey PLA instead of black. We should tinker a bit with it in different materials.

closed 2016-11-07 by Ghostkeeper