Ultimaker/CuraEngine #327: Issue When Infill Thickness /= Layer Height

CCS86, 2016-04-08:
I just attempted to slice a model with the 3/18 build, and set the layer height to 0.1mm, and the infill thickness to 0.2mm. The intention being to generate infill every other layer. What Cura generated is code that lays infill every layer.

Using Repetier to visualize, it doesn't seem that the extrusion amount is different for the infill (should be double the volume), the estimated time does not change, and the code is essentially identical, so it appears to be non-functional

[image]

The only difference in gcode between infill thickness = 0.1 vs 0.2:

[image]

BagelOrb, 2016-04-08:
The new Cura will be released very soon. Hopefully next week.
I can't reproduce the problem on the current version.

Perhaps it's a rounding problem; what happens if you fill in 0.20001?

CCS86, 2016-04-08:
No dice:

[0.2001 infill.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/210538/0.2001.infill.gcode.txt)

[0.2 infill.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/210535/0.2.infill.gcode.txt)

[0.1 infill.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/210536/0.1.infill.gcode.txt)

CCS86, 2016-04-18:
Bug still exists on the open beta build.

BagelOrb, 2016-07-12:
I've never reproduced this bug. Are _you_ still seeing it?

CCS86, 2016-07-12:
I think it was squashed in 2.1.2

Thanks!

closed 2016-07-12 by CCS86