Ultimaker/CuraEngine #328: Ooze shield going loose when printing

msl88, 2016-04-16:
Hi, 
I have problems with the "Ooze shield" going loose after some layers of printing, and this result in a lot of loose filament around the print. I'm printing with a self constructed printer and are using Repetier host V1.6.1 and the CuraEngine plugin following the Repetier Host.

I expirence the problem when I have a print with an angled shape that the shield tries to follow. I think the problem occurs because of the one thread thickness of the shield where the following layer do have a minimum of surface to attach to when it is printing in an angle e.x. at 45 degrees.

From my point of view I think there could be two solutions, which could be combined for the problem, if there not already exist a solution for this problem:
1. A printing option you can chose where the shield does not follow the shape of the part but instead print a "wall" in a 90 degrees angle in an approriate distance from the part.  

2. The possibility to chose the wall thickness of the ooze shield so it could be more than one thread of filament thick.

Hope you will look at my problem, and hope for a solution which can improve the ooze shield in the way I use it.

Please see the attached image for further explanation of the current problem.

[image]

Best regards 
Michael

Ghostkeeper, 2016-04-18:
Hmm, yeah I think option 1 would be the best option. Perhaps it even needs to be made default. It would require some thought on how to implement it efficiently though. We could flatten all vertices and take the convex hull of that, then offset the convex hull. Then we wouldn't need to compute the position of the shield for every layer any more.

MaximilianBuegler, 2018-07-13:
I would also like to see enhancements in the ooze shield. For me the fact that the ooze shield is printed with both extruders is causing the problem as I use different materials. Having an ooze shield extruder setting would be fine, so I can set the ooze shield to only be printed using my first extruder. I don't think 90° corners would solve anything, but an ooze shield line count would probably work. Same as increasing the ooze shield flow.

Ghostkeeper, 2018-08-27:
Option 1 is implemented now. It's called the Draft Shield. We'll see what we can do for option 2.

Ghostkeeper, 2018-08-28:
We don't think another ooze shield line is really necessary right now. We've never seen any trouble with the draft shield collapsing or falling over.

closed 2018-08-28 by Ghostkeeper