Ultimaker/CuraEngine #329: support area int not correct

zifengtayu, 2016-04-26:
[image]
[image]

the support erea isn't correct

zifengtayu, 2016-04-26:
version 2.1.0

BagelOrb, 2016-04-26:
Good catch!

Will fix this, hopefully soon. I'll have a chat with some of the frontend
guys.

What do you think is more intuitive: 90degrees = no support or 90 degrees =
support everywhere?

BagelOrb, 2016-04-26:
I can't reproduce this.

Is the overhang still in the wrong place if you trigger a reslice (adjust a setting and adjust it back)?

What operating system do you have?

What graphics card do you have?

zifengtayu, 2016-04-26:
I will send you the model,

zifengtayu, 2016-04-26:
ListElement { text: "0.1"}

zifengtayu, 2016-04-26:
I have send the model to your width email

BagelOrb, 2016-04-26:
Thank you.

The support is actually not in the wrong areas. The placement is just very suboptimal for this particular model.

Note that the frontend only colors faces red which violate the overhang angle, while the engine will also look at edges and vertices which violate the overhang angle. My guess is that the rounded edges have some small overhang pieces which get joined downward into one large piece of support.

In order to understand the different types of overhang I've attached an STL which exemplifies them. [overhang_point_line_plane_straight.stl.txt](https://github.com/Ultimaker/CuraEngine/files/236417/overhang_point_line_plane_straight.stl.txt) 

In order to solve your problems we would need to revise the support generation system in a big way. We currently don't have the time to do that. However, it remains to be seen how problematic the new support system is; if there are too many problems we might revise it in a short while after all.

Closing this issue, since we're not going to fix this issue specifically in its own.

closed 2016-04-26 by BagelOrb