Ultimaker/CuraEngine #330: CuraEngine Lay Flat command

sarahSamonte, 2016-04-26:
How can do the lay flat feature on CuraEngine? What is the terminal command for this?

Ghostkeeper, 2016-04-26:
There is none. It is implemented in the front-end. The back-end has no access to it.

Ghostkeeper, 2016-04-26:
To be precise, the code is here: https://github.com/Ultimaker/Uranium/blob/master/UM/Operations/LayFlatOperation.py#L46

It is not easy to call that with a command or to extract that to a separate script. It would have to do the model loading first. If I recall correctly though, that script was adapted from someone's C++ script on the Ultimaker forums. Perhaps we can find that back and see if that is easier to call from a command line.

nallath, 2016-04-26:
This script isn't an adaption of a c++ script, it's something that Daid cobbled together at some point in time. 
I did have contact with Christoph Schranz, who created a much better script (one that actually calculates a optimal print orientation. You can find it on https://github.com/iot-salzburg/STL-tweaker

closed 2016-04-26 by nallath

Ghostkeeper, 2016-04-26:
Ah, that's the one I thought of. In any case, that should be easy enough to call from a command line, and will probably help @sarahSamonte with her project.