Ultimaker/CuraEngine #331: The cover of clutter, the surface no retraction and z move

zifengtayu, 2016-04-28:
[image]
[image]

Ghostkeeper, 2016-04-29:
Looks good, right? When combing is activated it doesn't penetrate the wall any more, so the red line gets shifted over further back to go around. What is the problem we're looking at in these pictures?

zifengtayu, 2016-04-29:
the red line will destory the surface,need retraction and z hop when travel

BagelOrb, 2016-04-29:
Then don't enable combing, but enable retraction and zhop

closed 2016-04-29 by nallath

CCS86, 2016-04-29:
This gave me a thought:

Could we have a setting to specifically disable combing on skin surfaces, but leave it enabled everywhere else?

BagelOrb, 2016-04-29:
That's already implemented for the next release ^^