Ultimaker/CuraEngine #332: Heat bed temperature for rafts

Taiko1976, 2016-05-01:
**Setup:**
Renkforce RF2000 (dual extruder)
RepetierHost (v.1.6.1) on KUbuntu (Linux)
CuraEngine (v.3)

**Problem:**
When printing out rafts you may use different materials (e.g. PVA for raft and support structures, PLA for part). However PLA likes to be printed on a heat bed with 60°C while in my experience PVA works better on a head bed with 120°C. When printing PVA on a heat bed with 60°C it will not stick and will be all over the place instead of on the heat bed.

**Proposed Solution:**
The settings of CuraEngine provide some parameters for printing the raft. These settings should be extended by the temperature of the heat bed. Using this settings you can have a high temperature for the heat bed when printing the layers of the raft. After the raft is finished the head bed temperature can be lowered to the "normal" temperature".

Thanks and keep up your great work!
Regards, Bernd

BagelOrb, 2016-05-01:
I don't see a reason to lower the bed temperature after the raft has been printed. The stickiness would then only drop down and the raft would pop off more easily during printing, which leads to a failed print.

closed 2016-05-01 by BagelOrb

Ghostkeeper, 2016-05-02:
I suggest just keeping the bed at 120C for the whole print. As long as the PLA is more than 5mm away from the bed, it should have little effect on the PLA but still helps for the bed adhesion.