Ultimaker/CuraEngine #334: The surface is not delicate enough

zifengtayu, 2016-05-04:
[image]

[image]

zifengtayu, 2016-05-04:
the bottom one is sliced by CuraEngine

nallath, 2016-05-04:
The important question is; Does it even matter? Is there a noticable difference when you actually print it?

zifengtayu, 2016-05-04:
yes, The printing effect is very different

nallath, 2016-05-04:
Could you post those pictures? We can't read minds ;)

zifengtayu, 2016-05-04:
[image]
[image]

[image]

Ghostkeeper, 2016-05-04:
It looks to me like the top slicer actually tries to smooth low-polygon meshes explicitly. It looks like it creates G2/G3 commands to draw arcs.

BagelOrb, 2016-05-04:
CuraEngine does perform some simplification to optimize the slicing
computation times.

Can you post your model please?

I will try to find out whether this is still problematic in the current
beta release.

zifengtayu, 2016-05-04:
I send you by email 644748251@qq.com

zifengtayu, 2016-05-04:
In CuraEngine source code,which part i can change, to adjust the simplification, i can't find it.

BagelOrb, 2016-05-04:
Search for uses of Polygons::simplify(...)

BagelOrb, 2016-05-04:
Simplify will remove a point if it's less than 5 micron away from a direct line which skips the point...

Do note that the new beta release has many improvements over 15.04. It could very well be that Cura 2.1 beta doesn't have this problem. Please download it from the ultimaker software page and try it out.

BagelOrb, 2016-05-04:
This doesn't look too bad, right?
[image]

BagelOrb, 2016-05-04:
Hmm the old Cura actually looks better. Here's what 15.04 spits out:
[image]

BagelOrb, 2016-07-25:
How did you solve the problem in the end?

What code changes did you make?

Could you make a Pull Request out of those code changes?

BagelOrb, 2016-10-10:
This has been solved in Cura 2.3

closed 2016-10-10 by BagelOrb