Ultimaker/CuraEngine #337: Spiralized: Bug in Bottom Layers

CCS86, 2016-05-08:
In this gcode, it seems to get through the bottom layers, start spiralizing, then jump to printing non-spiral, multi-perimeter moves at layer 155:

[UM34_Doggy Vase-conc.gcode.txt](https://github.com/Ultimaker/CuraEngine/files/254052/UM34_Doggy.Vase-conc.gcode.txt)

[image]

[DoggyVase2.zip](https://github.com/Ultimaker/CuraEngine/files/254081/DoggyVase2.zip)

Ghostkeeper, 2016-05-09:
For reference, see also the discussion in https://github.com/Ultimaker/CuraEngine/issues/214.

BagelOrb, 2016-05-10:
Solving this ASAP.

BagelOrb, 2016-05-11:
There were some other issues which I solved along the way:
1) When there are multiple vases in one mesh, only one of the two layer parts would get spiralized
2) When you have compensate outer wall overlap enabled and there is a very thin piece in the vase, only the very last part of the outer wall would get spiralized.

closed 2016-05-11 by BagelOrb

CCS86, 2016-05-11:
Great, thanks!

Any idea on the date for the next release?

nallath, 2016-05-11:
In about 5-6 weeks (Depending on how well things go. We're doing some pretty major refactoring now).

Ghostkeeper, 2016-05-12:
Actually, this is fixed as a patch in version 2.1. It should be released when it goes out of beta.

CCS86, 2016-05-13:
Awesome, is that happening sooner?

BagelOrb, 2016-05-13:
Yes, in a week or two. ^^

Ghostkeeper, 2016-05-17:
Or whenever the marketing team is ready for it ;)