Ultimaker/CuraEngine #338: I want control fan when cross-sectional area small than a value

zifengtayu, 2016-05-09:
[image]

zifengtayu, 2016-05-09:
I can't make a sharp point.，I want to use a big fan，when cross-sectional area small than a value。

I do not modify the idea, if you can give me some tips.which code i should change on source code CuraEngine.

BagelOrb, 2016-05-09:
That's already implemented! :)

It's controlled by the maximum fan speed and the minimal layer time maximal
fan speed.

It's not dependent on the area, but on the rough layer time estimates.

Ghostkeeper, 2016-05-09:
The setting was renamed by the way. It's currently called "Regular/Maximum Fan Speed Threshold".

Basically you enter the amount of seconds for the threshold on the time it takes to print one layer, at which the fan speed is maximum. If you enter 5s, any layers that take less than 5s to print will get gradually stronger fans. For a sharp point such as that, the layer time is considerably less than 10s (which is the default), so the fan should get spinning. Unless that bowl on the left side of your screenshot is equally high, because then the nozzle moves away from the point and it gets time to cool down anyway. Basing the setting on time rather than polygon area is more logical if you ask me, since the cooling is also based on time rather than area.

For printing points such as that, the Lift Head setting is also useful. It moves the hot nozzle away from the print for a short while so it can solidify a bit, before it continues with the next layer.

closed 2016-05-09 by Ghostkeeper